stylesuxx/esc-configurator #22: Translation

DusKing1, 2021-03-11:
Can you port the translation to crowdin platform? I can do Simplified Chinese and Traditional Chinese.

stylesuxx, 2021-03-11:
That looks very interesting - will look into this, thank you.

stylesuxx, 2021-03-11:
Alright - I now started a public project -> https://crowdin.com/project/esc-configuratorcom Let me know if this is enough for you to start the translation, I have never worked with this tool before, so I am not sure. Will see how I can get this integrated into the app.

DusKing1, 2021-03-12:
Thank you! Never thought you're so fast reacted! Will do after this month. Might need you do something for me to get enough scope too.

stylesuxx, 2021-03-12:
No worries, whenever you are ready - really seems to be a cool system, it will probably make it easier to finde people who will contribute their language.

stylesuxx, 2021-04-13:
I am closing this since translations are set up. If anyone wants another language added, feel free to open a new issue or drop by on discord.

closed 2021-04-13 by stylesuxx