stylesuxx/esc-configurator #37: Allow flashing via FTDI adapter

stylesuxx, 2021-03-18:
This would allow ESC's to be flashed without a need for a flight controller as suggested by @AlkaMotors

One could simply use a FTDI adapter or arduino or anything similar.

stylesuxx, 2021-03-25:
This [library](https://github.com/dword1511/onewire-over-uart) allows to make a one wire interface from an Arduino and also shows how to convert a UART/FTDI adapter into a one wire interface.

knoopx, 2021-04-20:
I definitely need this for AM32, the configurator has this "direct connection" option but works on windows/linux only.

khari05, 2023-07-09:
Just wondering, is there any progress on this feature? I'm not sure if this is helpful but the original BLHeLiSuite supported flashing an Arduino with the correct firmware to interface over serial. A first step could just be adding support to use that same protocol to talk to an ESC after someone already sets up communication through BLHeLiSuite.

stylesuxx, 2023-07-09:
Nope, no progress on this one. It's a hard sell for me to be honest. I would need to dump a lot of time into this feature and I simply don't see the use case. FCs are relatively cheap, so if you buy an Arduino or an FC makes almost no difference. I know Alka is working on some additional bootloader functionality to integrate a 4 way interface. Once this is done I might look into it to at least allow AM32 being flashed via FTDI.

The Arduino 4 way interfaces I have seen all don't seem to work straight forward. And I can't really use the one that BLHeli ships with.

But by any means, if anyone wants to integrate this, I'd be more than happy to merge a PR.

khari05, 2023-09-06:
If anyone else finds this issue, the workaround I used was installing betaflight on an STM32 dev board and using that as a temporary flight controller that could talk to bluejay configurator