stylesuxx/esc-configurator #77: Motor control explanation

stylesuxx, 2021-04-03:
Motor control needs explanation:
* Motor control will not work if bi-directional DSHOT is  enabled but the ESC firmware does not support it, eg when flashing back from Bluejay to BlHeli
* Motor control will probably be at negativ max when user has 3d mode enabled (this needs to be tested though)

mathiasvr, 2021-04-03:
0 throttle (1000) is still a power down for 3d mode, but 1500 is idle and I think 1001 is negative max.

stylesuxx, 2021-04-03:
OK, I have hoped that it needs to reach idle before motors start spinning... Need to try it out and see what happens...

mathiasvr, 2021-04-03:
I haven’t really used/tested 3d mode a lot though, but actually a good idea in any case. I think we should make bluejay not start motors before idle is reached if that’s not already the case. But iirc it’s considered armed and ready to go as long as it sees 0 throttle/rc pulse, which I believe is the case even when fc is disarmed. However it could be that betaflight already handles this?

stylesuxx, 2021-04-03:
Yes, BF handling it, was also my hope, but I am not sure - guess there is only one way to find out...

stylesuxx, 2021-04-08:
Alright, so BF sets the motor sliders to center position when 3D mode is active. Which we should do here too, since it will spin full throttle once you move one of the sliders and then for example disable motor control (setting it to 0) - which is far from optimal. So just having a note is not enough here - 3D mode needs to be properly detected and the sliders default position needs to be set accordingly.

closed 2021-04-10 by stylesuxx