stylesuxx/esc-configurator #80: Install button/screen

mathiasvr, 2021-04-04:
The way you can install the pwa so it acts more like a native app can be easy to miss.

I think there should be a button on the main screen, or maybe even a welcome screen that only allows `install` or `use online`.
I think this will be great for people that prefer an installed app but do not necessarily know this is possible.
I even used the website for a long period simply because I wasn't approached directly with the option, even though I knew how to install it ;)

stylesuxx, 2021-04-04:
You are right. On Android they at least show you the option to install the first time you visit.
But since this is only displayed once and also only on mobile it might be easy to miss.

Welcome screen might be to intrusive, but maybe this could be solved via toast. Or maybe place it on home in the banner section...

mathiasvr, 2021-04-04:
IMO a welcome screen would not be intrusive, but I guess it depends on the user. It could guide the user before even loading the app at all. I think more might install it if it's not already loaded up but I'm not sure.

Noctaro, 2021-04-04:
To be honest, as user i would prefer a link in the disclaimer instead of a overlay that i have to click away.

stylesuxx, 2021-04-04:
The question is - what be a good motivator for the user to install it? The biggest "selling point" is offline usage - at least to some  extend. Once the user has flashed a firmware, it gets stored in the local storage and he can flash it without the need of downloading it again.

Next to the disclaimer is not prominent enough for me, welcome screen to intrusive. I am leaning towards putting it into the green area on home or to inform the user with a toast message - and telling him why he might be interested in installing it.

Noctaro, 2021-04-04:
Ok, i think if we need to draw more attention to it, the green area would be a good option. I dont know if the toast message would catch even more attention, but i would say lets give it a try and check how it feels.

closed 2021-04-07 by stylesuxx