stylesuxx/esc-configurator #83: [Feature Request] Add expert mode, add presets, and hide some parameters

DusKing1, 2021-04-04:
I'm thinking about AM32.

In general, I think settings like Sinusoidal Startup, complementary PWM, stuck protection, stall protection etc.. and maybe more settings should be set to invisible to general users. We should keep settings seems **SIMPLE** to general users. And when they wanna to change those settings, they should be in a level with not that bad understanding of what those parameters do in ESC firmware.

I'm also think add some preset button, for "RACE", "Freestyle", "Default", "X-Class" etc.. Maybe only those 4 are enough.

Also, @AlkaMotors ,those servo settings, once they applied then the drone settings will be overwrite right? How about add a checkbox or button to organize those servo/RC car settings?

Maybe this can also be applied on Bluejay? @mathiasvr what do you think?

stylesuxx, 2021-04-04:
I am not a big fan of hiding stuff from the user to be honest. But yeah, technically we could add a category field to the settings and then arrange them in an accordion or something similar.

About presets I have been thinking too, but I abandoned that Idea because I thought the setups and needs are so different that you can't really easily wrap them into a preset (or there would be so many presets, that it gets overwhelming quickly - that's the main reason why I added the hints, hoping they give enough information for the user to decide on their own). But technically this would not be too big of a  deal to implement.

Presets would also need to be maintained on a per layout basis - increasing the maintenance and testing work. For example: I don't have an X-Class, so who is going to test that preset? I don't want to offer presets of which I am not confident that they will work.

I would much rather prefer notes or hints:
> If you have an X-Class you might want to change settings for x and y because...

DusKing1, 2021-04-04:
We have some 12s X-class and X8 drones, and various type of big motors.

I come up this requests for several reasons:
- in the future we might have more and more settings, sliders, checkboxes etc. Some settings they have a relatively low using/setting frequency on most users. Put them there and force users look down or even scroll down to find those settings that they need to set/change/fiddling with is so inconvenient.
- also, on andriod, those many settings really forced users to scroll down to find the "Direction Reversed" checkbox.
- And, I'm sure we are those users who really cares what we should set/use, but many users (I wanna say 90%) just wanna press a sweet button and plug out and fly. In the history, those settings combinations were "taught" by those youtubers, bloggers etc, but this way also caused some big misleading and hatred to devs.

DusKing1, 2021-04-04:
Lemme elaborate my thoughts about "RACE", "Freestyle" presets.

Only change the motor timing, PWM frequency setting, startup power to a relatively robust setting.

- Freestyle: Variable PWM Frequency on, 22.5 degrees, 105 start power
- RACE: 24kHz fixed PWM frequency, 22.5 degress, 115 start power

And leave the motor-specific parameters - KV, poles - out to let users set them manually at their own.

mathiasvr, 2021-04-04:
I wanted to add settings categories to Bluejay configurator but never got around to it because I have been delaying all UI related changes. So for now the settings are only grouped in my notes. 😅 

I think presets might be difficult to get right, but only changing some parameters could be a good idea. But maybe it isn't so useful for Bluejay because there aren't so many settings to tune? In any case, I would let someone more skilled than me define the presets 😉

stylesuxx, 2021-04-04:
>  In any case, I would let someone more skilled than me define the presets

Same. Would be cool if we could get @spatzengr on board to provide some defaults. I trust him the most when it comes to tuning...

And yeah, categories are quickly implemented.

DusKing1, 2021-04-12:
Maybe I made some way too aggressive suggestions, I can see there are also a lot of ppl don't like hiding settings. Maybe add more tabs to organize settings better?

stylesuxx, 2021-04-12:
I can see how it makes sense to add Categorization - but I would like to hear from the firmware maintainers how they would like to categorize it, since they know their features best. AM32 has a lot of settings that are relevant for RC cars, so I could see how this could be moved in its own section.

For Bluejay/BlHeli I could only see two categories right now:
* Audio (Beep Strength, Beacon Strength, Beacon Delay)
* General (everything else)

But without knowing in Detail how @mathiasvr and @AlkaMotors have envisioned their settings to be nested, it makes little sense for me to implement anything. Eg.: Would a flat hierarchy be enough

* Category 1
* Category 2

or would nestings like so make sense:

* Category 1
  *  sub category 1.1

* Category 2
  *  sub category 2.1

Should those categories have any other fields, like for example collapsible/collapsed/open...

Also I think, the less clicks the user has to make, the better - that's why I am hesitant with accordions and such. I do understand that it would make sense on mobile though.

mathiasvr, 2021-04-12:
I think it's possible to divide the bluejay settings into more categories, but I would prefer a flat hierarchy.
However, I'm mostly interested in grouping the settings according to how advanced they are to give the user an indication of which settings they can try to adjust without much knowledge/risk vs which they should probably leave by default.
This type of grouping isn't really compatible with also having other categories as I see it and I don't know if there's an easy way to support both.

stylesuxx, 2021-04-12:
If a flat hierarchy is enough, then basically both variants can be supported. Each group/category would have a title and a matching description:

## Basic settings
No risk of destroying anything, feel free to play around here.
* settings
* ....

## Intermediate settings
You should at least somewhat understand what you are doing.
* settings
* ....

## Advanced settings
If you don't know what you are doing, you will set your house on fire.
* settings
* ....

Intermediate and advance could then also have the collapsed property. So Basci would always be visible and the other two you would need to expand to make adjustments.

DusKing1, 2021-04-15:
# Drone mode

## General
- Freestyle or Race mode button
- poles
- kv

## Advanced
- PWM frequency
- motor timing
- startup power
- ...

## Calibration
- Servo min
- Servo neutral
- Servo max
- ...

# Car mode

## Calibration
- Servo min
- ...

## Advanced
- Reversing style
- ...

What do you think? @AlkaMotors

AlkaMotors, 2021-04-16:
I am no UI expert. When it comes to how things are organized in the config tools that's up to you guys.

stylesuxx, 2022-10-23:
I am picking up this issues since it seems to start to make sense for some AM32 settings:
* Sine mode
* variable brake strength (needs on the fly dead-time adjustments)

Both of those settings might damage your ESCs if you are not sure what you are doing. Until expert settings are implemented, I will refrain from adding those AM32 settings (also see #249, #271).