stylesuxx/esc-configurator #88: Pre commit hook taking too long

stylesuxx, 2021-04-05:
Accoriding to [this comment](https://github.com/stylesuxx/esc-configurator/pull/86#issuecomment-813125283) the pre commit is taking too long.

@mathiasvr is it really the linter that takes so long, or rather the tests? Linter is done for me in under 5 seconds and the tests are done in under 8. Could you maybe post the error messages? I could move linting and testing to the run before pushing, but I actually like having the confidence to only commit tested stuff.

you can always commit with --no-verify to skip the checks.

mathiasvr, 2021-04-05:
Ah right it could be the tests running. In vscode there was no indication that the commit had “begun”, so I ended up starting the precommit over and over.

I think it’s good to have the linting on precommit, however the tests I think would be better on prepush, but that’s up to you.

I have a strange workflow where I commit a lot of different things and reorder it later, but that means it’s not always working before I’m done ;-)

stylesuxx, 2021-04-05:
Yes, I think I will set it up like this since I have the tests watching all the time anyway.

commit 3134c980 (2021-04-05): Run test pre-push instead of pre commit. Resolves #88
closed 2021-04-07 by stylesuxx