stylesuxx/esc-configurator #94: Some melodies

stylesuxx, 2021-04-08:
By r0b985:
```
ei65_bl_m:b=128,o=5,d=1:4a,8a#,8d,8g,8a#,8c6,8f,8a,4a#,8g,8a#,8d6,8d#6,8g,8d6,8c6,8a#,8d,8g,8a#,8a,8f,8a,1g,8g
ei65_bl_b:b=128,o=4,d=1:4p,2g,2f,2d#,2c,2g,4f.,1d#,8d#
```

```
TS_HOF_M:b=85,o=5,d=1:16a#5,16a5,16a#5,16g5,16a#5,16d5,16a#5,16g5,16a#5,16d#5,16d#6,16g5,16a#5,16d#5,16d#6,16a#5,16d6,16a#5,16d#6,16a#5,16f6,16a#5,16c6,16a5,16f6,16f5,16f6,16a#5,16c6,16f5,16f6,16a5
TS_HOF_B:b=85,o=3,d=1:8g,8d4,8g4,4d#,8a#,4d#4,8a#,8f4,8a#4,4f,8c4,4f4
```

stylesuxx, 2021-04-10:
Added as presets.

stylesuxx, 2021-04-13:
Tristam - Till It's Over:

> tris_tio_m:b=110,o=5,d=1:8a#,16d,8a,16d,4g,8d,16a,8d,16g,8f,16a#4,8g,16a#4,8g,16p,16g,8a#4,16a,8a#4,16a#,8g,16a#4,8d#,16a#4,8g,16a4,16c,8d,16a#,16p,16a#,16a,8g,16d,8a,16d,8a#.,16a,8d,8f,8g
tris_tio_b:b=110,o=4,d=1:2a#,2a,2g.,8g,8p,2d#,2f,g

Mozart:
> wam_trk_m:b=116,o=5,d=1:16h4,16a4,16g#4,16a4,4c,16d,16c,16h4,16c,4e,16f,16e,16d#,16e,16h,16a,16g#,16a,16h,16a,16g#,16a,4c6,8a,16c6,32f#,32a,8h,8a,8g,16a,32f#,32a,8h,8a,8g,16a,32f#,32a,8h,8a,8g,8f#,4e

> wam_trk_b1:b=116,o=3,d=1:4p,8a,16e4,16p,16e4,16p,16e4,16p,8a,16e4,16p,16e4,16p,16e4,16p,8a,16e4,16p,8a,16e4,16p,8a,16e4,16p,16e4,16p,16e4,16p,8e,16e4,16p,16e4,16p,16e4,16p,8e,16e4,16p,16e4,16p,16e4,16p,8e,16e4,16p,8h,8d4,4e

> wam_trk_b2:b=116,o=3,d=1:4p,8a,16c4,16p,16c4,16p,16c4,16p,8a,16c4,16p,16c4,16p,16c4,16p,8a,16c4,16p,8a,16c4,16p,8a,16c4,16p,16c4,16p,16c4,16p,8e,16h,16p,16h,16p,16h,16p,8e,16h,16p,16h,16p,16h,16p,8e,16h,16p,8h,8d4,4e

Rick Astley - Never gonna give you up:

> Never:d=4,o=3,b=210:8g3,8a3,8c4,8a3,4e4,8p,4e4,8p,4.d4,4p,8p,8g3,8a3,8c4,8a3,4d4,8p,4d4,8p,4c4,8b3,4.a3,8g3,8a3,8c4,8a3,2c4,4d4,4b3,4a3,4.g3,8p,4g3,2d4,2.c4,4p

stylesuxx, 2021-04-15:
smw_ath_m:b=165,o=5,d=1:16e6,8p.,16c6,8p,16g,16a,16c6,16p,16c6,8p.,16a,16g,16p,16c6,16p,16c6,16p,16g6,16p,16e6,8p,16d6,8p.,16g,16e6,8p.,16c6,8p,16g,16a,16c6,16p,16c6,8p.,16a,16g,16p,16c6,16p,16f6,16e6,16d6,8c6. 

smw_ath_b1:b=165,o=5,d=1:16c,16p,16g,16p,16g4,16p,16g,16p,16f4,16p,16f,16p,16f#4,16p,16f#,16p,16e4,16p,16e,16p,16d#4,16p,16d#,16p,16d4,16p,16f,16p,16g4,16p,16g,16p,16c,16p,16g,16p,16g4,16p,16g,16p,16f4,16p,16f,16p,16f#4,16p,16f#,16p,16d4,16p,16a,16p,16g4,16p,4g

smw_ath_b2:b=165,o=5,d=1:16c,16p,16e,16p,16g4,16p,16e,16p,16f4,16p,16c,16p,16f#4,16p,16c,16p,16e4,16p,16c,16p,16d#4,16p,16c,16p,16d4,16p,16d,16p,16g4,16p,16f,16p,16c,16p,16e,16p,16g4,16p,16e,16p,16f4,16p,16c,16p,16f#4,16p,16c,16p,16d4,16p,16f,16p,16g4,16p,4f

stylesuxx, 2021-04-16:
Vanessa Carlton - 1000 Miles:
vc_1000m_m:b=94,o=5,d=16:h4,h,h4,h,a#,h4,a#,4f#,8p,8d#,c#,h4,h,h4,h,a#,h4,a#,f#,h4,f#,h4,f#,h4,d#,e,d#,h4
vc_1000m_b:b=94,o=4,d=8:16p,e.,f#.,4d#,4p.,e.,f#.,4d#,4p.

ABBA - Chiquitita:

ab_chiq_m1:b=104,o=5,d=16:8f#,c#,p,c#,p,c#,p,c#,p,c#,p,c#,h4,c#,d,8f#,e6,e,e6,p,e6,e,e6,e,e6,e,e6,e,e6,e,g#6,g#,g#6,g#,g#6,g#,g#6,g#,a6,a,a6,a,a6,a,a6,a,g#6,g#,g#6,g#,a6,a,a6,a,h6,h,h6,h,8a6
ab_chiq_m2:b=104,o=6,d=16:8f#,c#,p,c#,p,c#,p,c#,p,c#,p,c#,h5,c#,d,8f#,a5,p,a5,p,a5,p,a5,p,a5,p,a5,p,a5,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p,8d
ab_chiq_b1:b=104,o=4,d=16:a3,p,a3,p,e,p,a3,p,a3,p,a3,p,e,p,a3,p,a3,p,a3,p,e,p,a3,p,a3,p,a3,p,h3,p,c#,p,d,p,d,p,a,p,d,p,d,p,d,p,a,p,d,p,d,p,d,p,a,p,d,p,d,p,d,p,8a
ab_chiq_b2:b=104,o=4,d=16:a3,p,a3,p,a,p,a3,p,a3,p,a3,p,a,p,a3,p,a3,p,a3,p,a,p,a3,p,a,p,a,p,h,p,c#5,p,d,p,d,p,d5,p,d,p,d,p,d,p,d5,p,d,p,d,p,d,p,d5,p,d,p,d,p,d,p,8d5

stylesuxx, 2021-05-05:
spider_dance_0:b=115,o=5,d=16:8f6,8c6,8g#,8f,p,b,a#,p,a#,g,d#,e,p,c6,a#,g#,a#,c6,e,f,g#,f,e,f6,p,d#6,32c6,32a#,g#,32c#4,32p,16c#4,16a#4,32c#4,32p,32c#4,32p,16c#4,16a#4,32c#4,32p,32c#4,32p,16c#4,16f4,32c#4,32p,32c#4,32p,16c#4,16a#4,16c#4,4c4,8p,16c4,16p,2c4

spider_dance_1:b=115,o=5,d=16:8c6,8g#,8f,8c,p,f,d#,p,g,d#,a#4,c,32f4,32p,32f4,32p,32f4,32p,16f4,32g4,32p,32g4,32p,32g4,32p,16g4,32g#4,32p,32g#4,32p,32g#4,32p,16g#4,32d#4,32p,32d#4,32p,32d#4,32p,16d#4,8f,8a#4,8g,8a#4,8g#,8a#4,8a#,8a#4,8g#,8c#,8c6,8f,8a#,8g#,4g

spider_dance_2:b=115,o=4,d=32:f,p,16f,16c5,f,p,f,p,16f,d#,p,16d#,c#,p,16c#,b#,p,b#,p,16b#,c,p,c,p,16c,16p,16g#5,16f5,16d#5,16g5,16g#5,16g4,16g#4,16c5,16g#4,16g4,16a#5,4p,1p,8c6,8a#5,8f6,8c6,8e6,8c#6

spider_dance_3:b=115,o=4,d=32:16f3,16c,16a#3,16c,16f3,16c,16a#3,16c,16c#,16b#,16a#3,16p,16c,16p,16g3,16a#3,16f3,16c,16a#3,16c,16f3,16c,16a#3,16c,16g#3,16c,16a#3,16c,16g#3,16c,16a#3,16c,16c#,16g#3,16g3,16g#3,16c#,16a#3,16g#3,16a#3,16c#,16c,16a#3,16c,c#,p,16c#,16c,16c#,2p.,4c6

stylesuxx, 2021-05-05:
bongocat:d=4,o=5,b=560:1g#,2b,2g#,2b,32p,2b,1g#,1f#,2a#,2f#,2e
Bongo cat

stylesuxx, 2021-05-05:
Chopin - Revolutionary Etude (warning: LONG)

rev_etude_0:o=5,b=160,d=16:8p,g#6,g6,f6,d6,d#6,d6,b,g,g#,g,f,d,d#,d,b4,g4,g#4,g4,f4,d4,d#4,d4,c4,g3,c4,g3,c4,g3,c4,g3,1d6,2p.,8g#.,g,1f6

rev_etude_1:o=5,b=160,d=16:1b,2p.,8g#.,g,8b3,g#6,g6,f6,d6,d#6,d6,b,g,g#,g,f,d,d#,d,b4,g4,g#4,g4,f4,d4,d#4,d4,c4,g3,c4,g3,c4,g3,c4,g3,1d6

rev_etude_2:o=5,b=160,d=16:1g,2p.,8f.,32g.,64p,1g,2p.,8f.,32g.,64p,8p,g#6,g6,f6,d6,d#6,d6,b,g,g#,g,f,d,d#,d,b4,g4,g#4,g4,f4,d4,d#4,d4,c4,b3,g4,f4,d#4,d4,d#4,d4,c4,b3,a#4,g#4,g4

rev_etude_3:o=5,b=160,d=16:1f,2p.,8d#.,g,1f,2p.,8d#.,32g.,64p,8g,8p,1p,1p,p,f4,g4,f4,d#4,d4,d#,d,c,b4,c,b4,g#4,g4,g#4,g4,f4,d#4,f4,d#4,2c4

stylesuxx, 2021-06-22:
Loituma - Ievan Polkka

```
ievan_polkka_0:o=5,b=100,d=16:8a#4,32d#.,64p,32d#.,64p,16d#.,32p,d#,f,32f#,.64p,f#,d#.,32p,8d#,8f#,f,32c#.,64p,32c#.,64p,32c#.,64p,8c#,8f,8f#,d#.,32p,8d#,8p
ievan_polkka_1:o=4,b=100,d=8:d#,p,a#3,p,d#,p,a#3,p,a#3,p,c#,p,d#,p,a#3,p
ievan_polkka_2:o=5,b=100,d=8:p,32a#.,64p,32a#.,64p,p,a#,p,a#,p,a#,p,32g#.,64p,32g#.,64p,p,g#,p,a#,p,a#
ievan_polkka_3:o=5,b=100,d=8:p,32f#.,64p,32f#.,64p,p,f#,p,f#,p,f#,p,32f.,64p,32f.,64p,p,f,p,f#,p,f#
```

stylesuxx, 2021-07-11:
Batman Theme

```
b=180,o=4,d=32:8d#5,p,8d#5,p,8d5,p,8d5,p,8c#5,p,8c#5,p,8d5,p,8d5,p,4d#5,p,p,2d#5
b=180,o=4,d=32:8g#,p,8g#,p,8g#,p,8g#,p,8g#,p,8g#,p,8g#,p,8g#,p,4g#,p,p,2g#
b=180,o=4,d=32:8d#6,p,8d#6,p,8d6,p,8d6,p,8c#6,p,8c#6,p,8d6,p,8d6,p,4g#6,p,p,2g#6
b=180,o=4,d=32:8g#,p,8g#,p,8g#,p,8g#,p,8g#,p,8g#,p,8d#6,p,8d#6,p,4d#6,p,p,2d#6
```

stylesuxx, 2021-09-12:
Super Mario power up:
```
Powerup_1: d=2,o=2,b=960: 8c4, 8g3, 8c4, 8e4, 8g4, 8c5, 8g4, 8g#3, 8c4, 8d#4, 8g#4, 8d#4, 8g#4, 8c5, 8d#5, 8g#5, 8d#5, 8d4, 8f4, 8a#4, 8f4, 8a#4, 8d5, 8f5, 8d5, 8f5, 8a#5, 8f5
Powerup_2: d=2,o=2,b=960: 8c5, 8g4, 8c5, 8e5, 8g5, 8c6, 8g5, 8g#4, 8c5, 8d#5, 8g#5, 8d#5, 8g#5, 8c6, 8d#6, 8g#6, 8d#6, 8d5, 8f5, 8a#5, 8f5, 8a#5, 8d6, 8f6, 8d6, 8f6, 8a#6, 8f6
Powerup_3: d=2,o=2,b=960: 8c4, 8g3, 8c4, 8e4, 8g4, 8c5, 8g4, 8g#3, 8c4, 8d#4, 8g#4, 8d#4, 8g#4, 8c5, 8d#5, 8g#5, 8d#5, 8d4, 8f4, 8a#4, 8f4, 8a#4, 8d5, 8f5, 8d5, 8f5, 8a#5, 8f5
Powerup_4: d=2,o=2,b=960: 8c5, 8g4, 8c5, 8e5, 8g5, 8c6, 8g5, 8g#4, 8c5, 8d#5, 8g#5, 8d#5, 8g#5, 8c6, 8d#6, 8g#6, 8d#6, 8d5, 8f5, 8a#5, 8f5, 8a#5, 8d6, 8f6, 8d6, 8f6, 8a#6, 8f6
```

Pacman tunes:

`pacman:d=4,o=5,b=112:32b,32p,32b6,32p,32f#6,32p,32d#6,32p,32b6,32f#6,16p,16d#6,16p,32c6,32p,32c7,32p,32g6,32p,32e6,32p,32c7,32g6,16p,16e6,16p,32b,32p,32b6,32p,32f#6,32p,32d#6,32p,32b6,32f#6,16p,16d#6,16p,32d#6,32e6,32f6,32p,32f6,32f#6,32g6,32p,32g6,32g#6,32a6,32p,32b.6`

Shorter variant:

`pacmanshort:d=4,o=5,b=112:32b,32p,32b6,32p,32f#6,32p,32d#6,32p,32b6,32f#6,16p,16d#6,16p,32c6,32p,32c7,32p,32g6,32p,32e6,32p,32c7,32g6,16p,16e6,16p,32d#6,32e6,32f6,32p,32f6,32f#6,32g6,32p,32g6,32g#6,32a6,32p,32b.6`

stylesuxx, 2022-01-15:
```
Rush_yyz-1:d=4,o=5,b=160:16a4,16a4,16f#4,16a4,16f#4,16e4,8e4,2c#4,16a4,16a4,16f#4,16a4,16f#4,16e4,8c#4,8b.3,8c#.4,8f#3
Rush_yyz-2:d=4,o=5,b=160:16a5,16a5,16f#5,16a5,16f#5,16e5,8e5,2c#5,16a5,16a5,16f#5,16a5,16f#5,16e5,8c#5,8b.4,8c#.5,8f#4
Rush_yyz-3:d=4,o=5,b=160:16a5,16a5,16f#5,16a5,16f#5,16e5,8e5,2c#5,16a5,16a5,16f#5,16a5,16f#5,16e5,8c#5,8b.4,8c#.5,8f#4
Rush_yyz-4:d=4,o=5,b=160:16c#4,8p.,16a4,8p.,16c#4,8p.,16c#5,8p.,16c#4,8p.,16a4,8p.,16c#4,8p.,16c#5,8p.
```

stylesuxx, 2022-02-12:
Slayer - South of Heaven

`SouthHeaven:b=108,o=5,d=4:8d#3,8d#4,8a#4,8b4,d#,8b4,d,8a#4,8c#,8a4,8c#,c.,8d#3,8d#4,8a#4,8b4,d#,8b4,d,8a#4,8c#,8a
`
Slayer - Raining Blood

`RainingBlo:d=4,o=5,b=240:8d#3,8d#4,8a4,8a#4,a4,8f4,8a4,g#4,8e4,8g#4,g4,8d#3,8d#3,8d#3,8d#4,8a4,8a#4,a4,8f4,8a4,g#4,8e4,8g#4,g4,8d#3,8d#3,8d#3,8a#4,8e,8f,e,8c,8e,d#,8b4,8d#,d,8d#3,8d#3,8d#3,8a#4,8e,8f,e,8c,8e,d#,8b4,8d`

stylesuxx, 2022-02-16:
Professor Chunk
```
StarTrekVoyager:d=4,o=6,b=45:8c.,8f.5,8f.,16e.,16c.,16c.,16d.,16d.,16c.,8c.,8a#5,32f.5,32c.,32f.5,8f.5
 StarTrekTNG:d=32,o=6,b=63:8f#.5,b.5,d#.,8c#.,a.5,g#.,4f#.
Meglovania 4CH 
Melody:b=112,o=5,d=4:32d,32p,16d,16d6,16p,16a,8p,16g#,16p,16g,16p,8f,16d,16f,16g,32c,32p,16c,8d6,16a,8p,16g#,16p,16g,16p,8f,16d,16f,16g,32a#4,32p,16a#4,16d6,16p,16a,8p,16g#,16p,16g,16p,8f,16d,16f,16g,32a4,32p,16a4,8d6,16a,8p,8g#,8g,8f,16d,16f,16g
Melody:b=112,o=4,d=4:32d,32p,16d,8d5,16a,8p,16g#,16p,16f,16p,8f,16d,16f,16g,32c,32p,16c,8d5,16a,8p,16g,16p,16f,16p,8f,16d,16f,16g,32a#3,32p,16a#3,8d5,16a,8p,16g,16p,16f,16p,8f,16d,16f,16g,32a3,32p,16a3,8d5,16a,8p,16g,16p,16f,16p,8f,16d,16f,16g
Melody:b=112,o=3,d=8:d4,p,d4,p,d4,p,d4,p,c4,p,c4,p,c4,p,c4,p,a#,p,a#,p,a#,p,a#,p,a,p,a,p,a,p,a,p
Melody:b=112,o=3,d=8:a,p,a,p,a,p,a,p,g,p,g,p,g,p,g,p,f,p,f,p,f,p,f,p,e,p,e,p,e,p,e,p
XFiles:b=125,o=5,d=4:e,b,a,b,d6,2b.
2Pac - Hit Em Up : d=4,o=5,b=200:a,8p,a,8g,a,p,a,c6,a,d6,8p,d6,8c6,d6,2p,8p,8d6,8d#6,e6,8p,e6,8d6,e6,p,c6,8b,8g,e,a.,a,8g,a
```

Ryano12, 2022-08-15:
how are you getting the "multi-channel" style rtttl's generated?

Ryano12, 2022-08-15:
How are you getting the "multi-channel" RTTTL's generated?

stylesuxx, 2022-08-15:
They are all hand crafted, not generated.

cypher1591, 2022-09-03:
It's verry funny that someone already did megalovania because i just did aswell. His version is also great so someone has to do a hard decision.
I also made the Giorno's Theme

**My Megalovania:**
```
Megalo:b=280,o=5,d=8: 4d,d6,p,a,4p,g#,p,g,p,4f,d,f,g, 4c,d6,p,a,4p,g#,p,g,p,4f,d,f,g, 4b,d6,p,a,4p,g#,p,g,p,4f,d,f,g, a#,a#,d6,p,a,4p,g#,p,g,p,4f,d,f,g
Megalo:b=280,o=6,d=8:4d,d7,p,a,4p,g#,p,g,p,4f,d,f,g, 4c,d7,p,a,4p,g#,p,g,p,4f,d,f,g, 4b,d7,p,a,4p,g#,p,g,p,4f,d,f,g, a#,a#,d7,p,a,4p,g#,p,g,p,4f,d,f,g
Megalo:b=280,o=4,d=8:4d,d5,p,a,4p,g#,p,g,p,4f,d,f,g, 4c,d5,p,a,4p,g#,p,g,p,4f,d,f,g, 4b,d5,p,a,4p,g#,p,g,p,4f,d,f,g, a#,a#,d5,p,a,4p,g#,p,g,p,4f,d,f,g
Megalo:b=280,o=4,d=8:d,p,d,p,d,p,d,p,d,p,d,p,d,p,d,p, 4e,4p,4e,4p,4e,4p,4e,4p, f,p,f,p,f,p,f,p,f,p,f,p,f,p,f,p, 4g,4p,4g,4p,4g,4p,4g,4p
```

**Giorno's Theme**
```
Girono:d=8,o=6,b=140: 4f#,f#,4d,d,4p,   f,32f,32p,e,32e,32p,p,c#,32c#,32p,d,32d,32p,p, 4f#5,f#5,4b5,b5,4p,   d65,32d,32p,16p,4p,c#,32c#,32p,a5,16a5,g5
Girono:d=8,o=5,b=140: 4f#,f#,4d,d,d,e,   f,16f,e,16e,d,c#,16c#,d,16d,e, 4p,p,4d,d,b4,c#,   d,16d,e,16e,d,c#,16c#
Girono:d=8,o=4,b=140: 4b#,b#,4a,a,4p,   a,32a,32p,16p,a,32a,32p,16p,g#,32g#,32p,16p,g#,32g#,32p,16p, 4b,b,4a,a,4p, a,32a,32p,16p,   a,32a,32p,16p,  g#,32g#,32p,16p,  g#,32g#,32p,16p
Girono:d=8,o=4,b=140: 4f#,f#,4f#,f#,4p,   f,32f,32p,16p,f,32f,32p,16p,e,32e,32p,16p,e,32e,32p,16p, 4f#,f#,p,2p, f,32f,32p,16p,   f,32f,32p,16p,  e,32e,32p,16p,  e,32e,32p,16p
```

namuol, 2022-09-10:
Your tiny whoop is cool, but can it run DOOM? Now it can!

https://gist.github.com/namuol/c81333b1247c8deaa0d6a1664d419713

```
e1m1:b=420,o=4,d=32:8c4,8p,4c4,4c5,8c4,8p,4c4,4a#4,8c4,8p,4c4,4g#4,8c4,8p,4c4,4f#4,8c4,8p,4c4,4g4,4g#4,8c4,8p,4c4,4c5,8c4,8p,4c4,4a#4,8c4,8p,4c4,4g#4,8c4,8p,4c4,1f#4
e1m1:b=420,o=4,d=32:8c4,8p,4c4,4c5,8c4,8p,4c4,4a#4,8c4,8p,4c4,4g#4,8c4,8p,4c4,4f#4,8c4,8p,4c4,4g4,4g#4,8c4,8p,4c4,4c5,8c4,8p,4c4,4a#4,8c4,8p,4c4,4g#4,8c4,8p,4c4,1c4
e1m1:b=420,o=4,d=32:4c4,4p,4c4,4p,4c4,4p,4c4,4p,4c4,4p,4c4,4p,4c4,2c4,4p,4c4,2p,4p,4c4,2p,4p,4c4,4p,4c4,1c4
e1m1:b=420,o=4,d=32:8c5,8p,4c5,4c6,8c5,8p,4c5,4a#5,8c5,8p,4c5,4g#5,8c5,8p,4c5,4f#5,8c5,8p,4c5,4g5,4g#5,8c5,8p,4c5,4c6,8c5,8p,4c5,4a#5,8c5,8p,4c5,4g#5,8c5,8p,4c5,1f#4
```

arwebSE, 2022-12-06:
Hi! I was inspired by the included "Pokémon Red/Blue Title Theme" and made a very short  "Pokémon Red/Blue Opening Theme" that I feel fits well when booting an ESC =)

```
PkmnOpening1:d=8,o=7,b=198:16d4,16p,16d4,16p,4a4,16d4,16p,16d4,16p,4a#4,16d4,16p,16d4,16p,4c5,16d4,16p,16d4,16p,4c#5,1d5
PkmnOpening2:d=8,o=7,b=198:16a3,16p,16a3,16p,4d4,16a3,16p,16a3,16p,4d#4,16a3,16p,16a3,16p,4f4,16a3,16p,16a3,16p,4g4,1f#4
PkmnOpening3:d=8,o=7,b=198:16d4,16p,16d4,16p,4a4,16d4,16p,16d4,16p,4a4,16d4,16p,16d4,16p,4a#4,16d4,16p,16d4,16p,4a#4,1a4
PkmnOpening4:d=8,o=7,b=198:16d4,16p,16d4,16p,4a4,16d4,16p,16d4,16p,4a#4,16d4,16p,16d4,16p,4c5,16d4,16p,16d4,16p,4c#5,1d5
```

kilozebra, 2022-12-07:
The intro riff to "Jingle Bell Rock" by Bobby Helms:

```
jingle1:b=200,o=5,d=4:f#,32p,8f#,32p,8f#,p,f#,32p,8f#,32p,8f#,p,f#,32p,8a,32p,d,32p,8e,64p,f,32p,8e,32p,d,32p,8b4,32p,2a4
jingle2:b=200,o=5,d=4:d,32p,8d,32p,8d,p,c#,32p,8c#,32p,8c#,p,d,32p,8e,32p,a4,32p,8b4,32p,c,32p,8b4,64p,a4,32p,8g4,32p,2f#4
jingle3:b=200,o=5,d=4:f#,32p,8f#,32p,8f#,p,f#,32p,8f#,32p,8f#,p,f#,32p,8a,32p,d,32p,8e,64p,f,32p,8e,32p,d,32p,8b4,32p,2a4
jingle4:b=200,o=5,d=4:d,32p,8d,32p,8d,p,c#,32p,8c#,32p,8c#,p,d,32p,8e,32p,a4,32p,8b4,32p,c,32p,8b4,64p,a4,32p,8g4,32p,2f#4
```

stylesuxx, 2022-12-11:
Guns n Roses - Sweet Child O Mine
```
SweetChi:d=4,o=5,b=125:8f6,8g#,8d#,8c#6,8g#,8f#,8f#6,8g#,8f6,8g#,8d#,8c#6,8g#,8f#,8f#6,8g#,8f6,8g#
```
 
Violent Femmes - Blister In the Sun
```
BlisterI:d=4,o=5,b=285:e.,8g#.,8e.,a.,8g#.,e.,g#.,8e.,a.,g#.,e.,8g#.,8e.,a.,8g#.,e.,p,p,p,p,p,e.,8g#.,8e.,a.,8g#.,e.,g#.,8e.,a.,g#.,e.,8g#.,8e.,a.,8g#.,e.
```

Violent Femmes - Blister In the Sun (short)
```
BlisterI:d=4,o=5,b=285:e.,8g#.,8e.,a.,8g#.,e.,g#.,8e.,a.,g#.,e.,8g#.,8e.,a.,8g#.,e.
```

Kino - A Pack of Cigarettes
```
Kino1:b=128,o=7,d=8:b4,c5,b5,a5,e5,4c5,g5,e5,c5,4d5,a5,f#5,c5,b4,b4,b5,a5,e5,4c5,g5,e5,b4,c5,p,a5,f#5,c5,4b4
Kino2:b=128,o=7,d=8:e4,e3,4a3,p,a3,4c4,p,c4,4d4,p,d4,4e4,4e4,4a3,p,a3,4c4,p,c4,4d4,p,d4,4e4
Kino3:b=128,o=7,d=8:b4,c5,b5,a5,e5,4c5,g5,e5,c5,4d5,a5,f#5,c5,b4,b4,b5,a5,e5,4c5,g5,e5,b4,c5,p,a5,f#5,c5,4b4
Kino4:b=128,o=7,d=8:e4,e3,4a3,p,a3,4c4,p,c4,4d4,p,d4,4e4,4e4,4a3,p,a3,4c4,p,c4,4d4,p,d4,4e4
```

stylesuxx, 2023-01-03:
Star Wars - Cantina Song
`Cantina:d=4,o=5,b=250:8a,8p,8d6,8p,8a,8p,8d6,8p,8a,8d6,8p,8a,8p,8g#,a,8a,8g#,8a,g,8f#,8g,8f#,f.,8d.,16p,p.,8a,8p,8d6,8p,8a,8p,8d6,8p,8a,8d6,8p,8a,8p,8g#,8a,8p,8g,8p,g.,8f#,8g,8p,8c6,a#,a,g`

stylesuxx, 2023-01-03:
cheech and chong "earache my eye"

ESC1 + ESC3: 
`Earache:d=4,o=3,b=180:8b,8a#,a,8b,8a#,a,d4,32p,d4,1b,1b`

ESC2:
`Earache:d=4,o=4,b=180:8b,8a#,a,8b,8a#,a,d5,32p,d5,1b,1b`

ESC4:
`Earache:d=4,o=4,b=180:8f#,8f,e,8f#,8f,e,a,32p,a,1f#,1f#`

stylesuxx, 2023-02-01:
```
Snap! - Rythm Is A Dancer
motor1:b=280,o=5,d=16:p,4e,p,4c,p,4d,p,4c,p,4f4,p,4f4,p,4a4,p,4f4,p,4g4,p,4g4,p,4b4,p,4g4,p,4a4,p,4a4,p,4c,p,4a4
motor2:b=280,o=4,d=32:8a,p,8a,p,8a,p,8a,p,8a,p,8a,p,8a,p,8a,p,8f,p,8f,p,8f,p,8f,p,8f,p,8f,p,8f,p,8f,p,8g,p,8g,p,8g,p,8g,p,8g,p,8g,p,8g,p,8g,p,8a,p,8a,p,8a,p,8a,p,8a,p,8a,p,8a,8p
motor3:b=280,o=4,d=32:8a,p,8a,p,8a,p,8a,p,8a,p,8a,p,8a,p,8a,p,8f,p,8f,p,8f,p,8f,p,8f,p,8f,p,8f,p,8f,p,8g,p,8g,p,8g,p,8g,p,8g,p,8g,p,8g,p,8g,p,8a,p,8a,p,8a,p,8a,p,8a,p,8a,p,8a,8p
motor4:b=280,o=5,d=16:4e,p,4c,p,4d,p,4c,p,4f4,p,4f4,p,4a4,p,4f4,p,4g4,p,4g4,p,4b4,p,4g4,p,4a4,p,4a4,p,4c,p,4a4,p
```
```
WinXPStartup1:b=100,o=7,d=8:4f3.,4g#3.,4d#3
WinXPStartup2:b=100,o=7,d=8:4c4.,4d#4.,4a#3
WinXPStartup3:b=100,o=7,d=8:4g#4.,4g#4.,4d#4
WinXPStartup4:b=100,o=7,d=8:d#5,16d#4,a#4.,g#4,d#4,d#5,4a#4
```

```
The Chemical Brothers - Galvanize:
ESC1: Galvanize:d=16,o=3,b=104:8f,p,8f,p,8g,p,g5,4p,a#,8p,8a,8p,8g#
ESC2: Galvanize:d=16,o=4,b=104:8f,p,8f,p,8g,8p,g,p,a,p,c5,a#,p,a#,a,a#,g,a,f
ESC3:  Galvanize:d=16,o=3,b=104:8b,p,8b,p,8b,8p,g5,p,a5,p,c6,a#5,p,a#5,a5,a#5,g5,a5,f5
ESC4: Galvanize:d=16,o=3,b=104:8g,p,8g,p,8g,p,g5,4p,a#,8p,8a,8p,8g#
```

kilozebra, 2023-02-11:
Owl City - Fireflies:
```Fireflies1:d=8,o=5,b=200:a#3,a#4,d6,a#3,a#4,a#5,p,d#,d#4,a#,f,d#,f,a#,p,d#,g#3,c,a#,c,d#,d#4,g#3,p,a#,c,a#,c4,a#,d#,f,a#3,a#4,d6,d,f4,a#,p,d#,d#4,a#4,f,d#,g#,g,d#,a#4,g#3,c,d#4,g#3,a#4,p,d#4,c,g#3,c,a#4,c4,d#,f
Fireflies2:d=8,o=4,b=200:4p,a#3,f.,16p,4f,d#,4f,4d#,4a#3,4p,4p,a#3,c.,16p,4c,a#3,4c,4d#,4f.,4p,4g,4f,d#,a#3.,16p,16a#3.,32p,4a#3,f,4d#,4c,1p,2p
Fireflies3:d=8,o=4,b=200:4p,a#3,f.,16p,4f,d#,4f,4d#,4a#3,4p,4p,a#3,c.,16p,4c,a#3,4c,4d#,4f.,4p,4g,4f,d#,a#3.,16p,16a#3.,32p,4a#3,f,4d#,4c,1p,2p
Fireflies4:d=8,o=4,b=200:4p,a#3,f.,16p,4f,d#,4f,4d#,4a#3,4p,4p,a#3,c.,16p,4c,a#3,4c,4d#,4f.,4p,4g,4f,d#,a#3.,16p,16a#3.,32p,4a#3,f,4d#,4c,1p,2p
```

Shadow0624, 2023-03-06:
Pink Panther Theme:
ESC1:  PinkPanther:d=4,o=5,b=100:16c#4,8d4,32p,16d#4,1e4,8p,16d#4,8d4,32p,16c#4,1c4,16p
ESC 2+3+4 : PinkPanther:d=4,o=5,b=100:p,16d#4,e4,8p,16f#4,g4,16p,16d#4,8e4,64p,16f#4,8g4,64p,16c5,8b4,64p,16e4,8g4,64p,16b4,2a#4,16a4,16g4,16e4,16d4,e4,16p

GKLove, 2023-03-30:
I can't Defeat Airman
ESC 1&2&3&4
`Airman:b=118,o=4,d=16:e,g,a,b,p,8p,e,b,p,a,8p,b,p,a,g,f#,e,d,p,e,p,f#,p,a,p,a,g,f#,g,p,e,g,a,b,p,8p,e,b,p,a,8p,b,p,a,g,f#,e,d,p,e,p,f#,p,a,p,a,g,a,b`

Brain Power
ESC 1&2
`Brainpower:b=170,o=4,d=32:16e5,16p,16f#5,16p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,e5,p,e5,p,e5,p,e5,p,16b,16p,16c#5,16p,16a,16p,16f#5,16p,16f#5,16p,16e5`
ESC 3&4
`Brainpower:b=170,o=4,d=32:1p,1p,16e5,16p,16f#5,16p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,f#5,p,d5,p,d5,p,d5,p,d5,p,16f#5,16p,16a5,16p,16g#5,16p,16a5,16p,16g#5,p,16g#5,16c#5`

Unbreakable Determination
ESC 1&2&3&4
`ninjia:b=150,o=4,d=8:f#5,a5,f#5,b5,16e5,f#5,16e5,f#5,c#6,32f#5,32p,32f#5,32p,32f#5,32p,32f#5,32p,f#5,g#5,a5,g#5,f#5,p,g#5,f#5,b5,g#5,16f#5,g#5,16f#5,g#5,c#6,32g#5,32p,32g#5,32p,32g#5,32p,32g#5,32p,g#5,a5,b5,a5,g#5`

Lagtrain
ESC 1
`Lagtrain:b=74,o=5,d=16:b4,a4,b4,e,f#,p,b4,e,f#,p,b4,f#,g,p,e,b,a,p,g,p,f#,32g,32f#,e,d6,p,32d6,32p,32d6,32p,d6,p,32b,32p,b,p,b4,a4,b4,e,f#,p,b4,e,f#,32b4,32p,b4,32d6,32p,d6,p,d6,b,p,b,a,p,g,f#,p,e`
ESC 2
`Lagtrain:b=148,o=5,d=4:p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,p,16f#,8p,16p,16f#,8p,16p`
ESC 3
`Lagtrain:b=148,o=5,d=1:p,p,p,p,p,p,p,2p,16f#,16p,16e,16p,8e,4p`
ESC 4
`Lagtrain:b=74,o=5,d=1:p,p,p,p`

夜よるに駆かける yoru ni kakeru
ESC 1&2&3&4
`yorunikakeru:b=120,o=5,d=4:16a#4,16c,16d#,16a#,16c6,8a#,8g,32g,8c,8d#,8f,16g,16c,16d#6,16d6,16a#,8g,8a#,32a#,8c6,8a#,8g,8f,8d,16g,8f,8d#,32d#,8c,8b4,8g#,8g,8d,8f,16d#,8a#,16a#,8g#,8g`

Among Us Theme
ESC 1&2&3&4
`Amongus:b=95,o=5,d=4:c4,8c6,8d#6,8f6,8f#6,8f6,8d#6,c6,8c6,16a#,16d6,c6,16c6,8g3,c4,8c6,8d#6,8f6,8f#6,8f6,8d#6,2f#6,16f#6,16f6,16d#6,16f#6,16f6,16d#6`

GKLove, 2023-04-02:
Bad Apple
ESC 1
`Melody:b=147,o=5,d=8:d#,f,f#,g#,16a#,p,16a#,d#6,c#6,16a#,p,16a#,16d#,p,16d#,1p,1p,2p,d#,f,f#,g#,16a#,p,16a#,d#6,c#6,16a#,p,16a#,16d#,1p,1p,4p`
ESC 2
`Melody:b=147,o=5,d=8:1p,2p,a#,g#,f#,f,d#,f,f#,g#,16a#,p,16a#,g#,f#,f,d#,f,f#,f,d#,d,f,1p,2p,a#,g#,f#,f,d#,f,f#,g#,16a#,p,16a#,1p,p`
ESC 3
`Melody:b=147,o=5,d=8:d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,b3,f#4,b3,f#4,b3,f#4,b3,f#4,c#4,g#4,c#4,g#4,d4,a#4,d4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,d#4,a#4,b3,f#4,b3,f#4,b3,f#4,b3,f#4,c#4,g#4,c#4,g#4,d4,a#4`
ESC 4
`Melody:b=147,o=5,d=8:1p,1p,1p,1p,1p,1p,2p,4p,g#,f#,16f,p,16f,16f#,p,16f#,16g#,p,16g#,16a#,p,16a#`

kilozebra, 2023-04-06:
Battle of the Holy (Castlevania Adventure)
```
battle1:b=120,o=5,d=16:e,b4,e,b,b4,e,a#,b4,e,a,b4,e,g,b4,e,b4,g,b4,e,a,b4,e,g,b4,e,f#,b4,e,f#,b4,b3,8a,32a,32p,8a#,32a#,32p,8b.
battle2:b=120,o=4,d=16:1e3,2c.,8d,8p,8e3,8p,8e3,8p
battle3:b=120,o=4,d=16:1p,2p.,b3,b5,8e3,8p,8e3,8p,8e3
battle4:b=120,o=5,d=16:e,b4,e,b,b4,e,a#,b4,e,a,b4,e,g,b4,e,b4,g,b4,e,a,b4,e,g,b4,e,f#,b4,e,f#,b4,b,8a,32a,32p,8a#,32a#,32p,8b.
```

Peter Gunn Theme (Spy Hunter, Blues Brothers)
```
pgunn1:d=4,o=3,b=260:8f.,16p,f,g,f,8g#,8a,f,a#,a,8f.,16p,f,g,f,8g#,8a,f,a#,a,8f.,16p,f,g,f,8g#,8a,f,a#,a,8f.,16p,f,g,f,8g#,8a,f,a#,a
pgunn2:d=4,o=5,b=260:1d#,2d#.,c,1p,1p,1d#,1d#,c6.,f#.,8f,8e,8d#,8d,8c#,c,p.
pgunn3:d=4,o=4,b=260:8f.,16p,f,g,f,8g#,8a,f,a#,a,8f.,16p,f,g,f,8g#,8a,f,a#,a,8f.,16p,f,g,f,8g#,8a,f,a#,a,8f.,16p,f,g,f,8g#,8a,f,a#,a
pgunn4:d=4,o=4,b=260:1d#,2d#.,c,1p,1p,1d#,1d#,c5.,f#.,8f,8e,8d#,8d,8c#,c,p.
```

Lordnibbler, 2023-04-10:
In case anyone wants to troll, this is a pretty good take on the DJI Mavic/Phantom 4 startup chime

```
DJI: d=2,o=2,b=295: 4c6, 4d6, 4g6, 16p, 4d6, 4g6, 16p, 4d6, 4g6
DJI: d=2,o=2,b=295: 4c6, 4d6, 4g6, 16p, 4d6, 4g6, 2p, 164p, 4g6, 16p, 4d6, 4g6
DJI: d=2,o=2,b=295: 4c6, 4d6, 4g6, 16p, 4d6, 2p, 4p, 16p
DJI: d=2,o=2,b=295: 4c6, 4d6, 4g6, 1p, 8p
```

GKLove, 2023-04-10:
Megaman2 Dr.WILY Stage 1
ESC 1
`Melody:b=189,o=5,d=16:c#,p,32e,32p,32e,32p,32e,32p,p,32e,32p,32e,32p,32e,32p,p,c#,p,8p,c#,p,32e,32p,32e,32p,32e,32p,p,e,p,c#,p,8p,g#,p,f#,p,g#,1p,1p,1p,1p`
ESC 2
`Melody:b=189,o=5,d=16:1p,1p,8p,32e,32p,32e,32p,32e,32p,p,32e,32p,32e,32p,32e,32p,p,c#,p,8p,g#5,p,8p,f#5,p,8p,e5,p,8p,f#5,1p,1p,1p,1p`
ESC 3
`Melody:b=189,o=5,d=16:1p,1p,1p,1p,32f#5,32p,32f#5,32p,32f#5,32p,p,32f#5,32p,32f#5,32p,32f#5,32p,p,d#5,p,8p,g#5,p,8p,f#5,p,8p,e5,p,8p,d#5,1p,1p,1p`
ESC 4
`Melody:b=189,o=5,d=16:1p,1p,1p,1p,1p,2p,4p,c#5,p,c#5,p,c#5,p,g#5,p,b5,p,a#5,p,4p,c#5,p,c#5,p,c#5,p,g#5,p,b5,p,a#5,p,8p,d#6,p,e6`

Airwolf359, 2023-04-10:
DukesOfHazzardHorn:b=336,o=5,d=32:8g.,p,8e.,p,4c,p,4c,p,8c.,p,8d.,p,8e.,p,8f.,p,4g,p,4g,p,4g,p,4e

RideOfTheValkyries:b=285,o=5,d=8:b4,e,b4,e,4g.,4e.,g,e,g,4b.,4g.,b,g,b,4d6.,4d.,g,d,g,4b.

GKLove, 2023-04-12:
Monkeys Spinning Monkeys
ESC 1
`Melody:b=142,o=5,d=16:g,p,8p,g#,p,a,p,8p,g#,p,8p,f#,p,g,p,8p,d#,p,e,p,2p,g,p,8p,g#,p,a,p,8p,g#,p,8p,f#,p,e6,p,8p,b,p,c6,2p,p`
ESC 2
`Melody:b=142,o=5,d=16:1p,1p,1p,2p,g,p,e,p,d,8p`
ESC 3
`Melody:b=142,o=5,d=16:8p,c,p,8p,c,p,8p,c,p,8p,c,p,8p,c,p,8p,c,p,g4,p,a4,p,8p,c,p,8p,c,p,8p,c,p,8p,c,p,8p,c,p,8p,c,p,8p,c,p,2p`
ESC 4
`Melody:b=142,o=5,d=16:1p,1p,1p,2p,g3,p,a3,p,b3,p,c4`

GKLove, 2023-04-13:
Cirno's Perfect Math Class
ESC 1
`Melody:b=180,o=5,d=8:a,a#,b,c6,p,c6,b,a#,a,a#,b,c6,p,c6,b,a#,a,a#,b,c6,d6,e6,f6,d6,g6,f6,e6,c6,p,16c,32c,32p,4c,1p,1p,1p,2p`
ESC 2
`Melody:b=180,o=5,d=8:a4,a#4,b4,c,p,c,b4,a#4,a4,a#4,b4,c,p,c,b4,a#4,f4,g4,g#4,a4,a#4,c,d,a#4,e,d,c,g4,p,16b4,32b4,32p,4b4,1p,1p,1p,2p`
ESC 3
`Melody:b=180,o=5,d=8:1p,1p,1p,1p,a,a#,b,c6,p,c6,b,c6,d6,e6,f6,a6,p,a6,g6,a6,a#6,a6,a#6,g6,e6,d6,e6,c6,16f6,32f6,32p,f6,e6,f6`
ESC 4
`Melody:b=180,o=5,d=8:1p,1p,1p,1p,a4,a#4,b4,c,p,c,b4,c,4p,p,d,p,d,e,f#,4g,g,16g,32g,32p,2g,a,g#,g,f`

Noctaro, 2023-04-16:
Super Mario Intro:d=4,o=5,b=120:16e6,16e6,16p,16e6,16p,16c6,16e6,16p,16g6,16p,16p,16p,16g5,16p,16p,16p

GKLove, 2023-04-18:
Devil's Swing
ESC 1&2&3&4
`Melody:b=125,o=5,d=16:a#6,p,f#6,p,32f#6,32p,32f#6,32p,f#6,d#6,f#6,g6,p,g#6,8p,f#6,d#6,g#6,p,f#6,p,f#6,p,f#6,d#6,32f#6,32p,f#6,p,f#6,8p,f#6,d#6,f#6,p,f#6,p,32f#6,32p,32f#6,32p,f#6,d#6,f#6,g#6,p,a6,8p,a6,p,a6,p,a6,p,a6,p,g#6,p,a6,g#6,p,f#6`

Rage Your Dream
ESC 1&2&3&4
`Melody:b=120,o=5,d=8:g,a#,c6,g4,c6,d#6,c6,g#4,d#,d#6,d6,a#4,a#5,g5,a#5,g4,g#5,a#5,c6,a#4,d#5,c6c,f5,c4,d#5,f5,g5,a#4,d6,16a#5,32a#5,32p,a#5,d5`

Merry Christmas Mr. Lawrence
ESC 1&2&3&4
`Melody:b=96,o=5,d=8:d#5,f5,d#5,a#4,2d#5,d#5,p,d#5,f5,d#5,f5,g#5,f5,d#5,f5,d#5,a#4,2c#5,c#5,p,4c#6,c6,g#5,4f5,d#5,f5,d#5,a#4,2d#5,d#5,p,d#5,f5,d#5,f5,g#5,f5,d#5,f5,d#5,c#5,2a#4`

Gourment Race
ESC 1&2&3&4
`Melody:b=176,o=5,d=16:g5,d#5,c5,g#4,g4,g#4,c5,d#5,f5,d5,a#4,a4,f4,a4,a#4,d5,d#5,c5,g#4,g4,d#4,f4,g4,c5,f5,d5,a#4,g4,f4,g4,a#4,d5,d#5,c5,g#4,g4,d#5,c5,g#4,g4,d5,a#4,a4,f4,d5,a#4,a4,f4,c5,a4,f4,d4,c5,a4,f4,d4,g5,p,8p,g5`

BONES
ESC 1&2&3&4
`Melody:b=114,o=5,d=8:a#4,c#5,a#4,c#5,4f5,d#5,c#5,16d#5,32d#5,32p,16d#5,32d#5,32p,16d#5,32d#5,32p,16d#5,32d#5,32p,4f5,d#5,c#5,16d#5,32d#5,32p,16d#5,32d#5,32p,16d#5,32d#5,32p,16d#5,32d#5,32p,4f5,d#5,c#5,16d#5,32d#5,32p,16d#5,32d#5,32p,16a#4,32a#4,32p,16a#4,32a#4,32p,2c#5`

GKLove, 2023-04-21:
Bonetrousle

ESC 1&2
`Melody:b=150,o=5,d=16:c5,p,g5,p,c6,p,d#6,p,8p,c6,p,8p,b5,p,c6,p,d6,p,8p,c6,p,g5,p,d#5,p,c5,p,d5,p,d#5,p,f#5,p,8p,g5,p,d#5,p,c5,p,g4,p,b4,p,c5,p,d5,p,8p,c5`
ESC 3
`Melody:b=150,o=5,d=16:8p,d#4,p,8p,f#4,p,8p,d#4,p,8p,d4,p,8p,d#4,p,8p,f#4,p,8p,d#4,p,8p,d4,p,8p,d#4,p,8p,f#4,p,8p,d#4,p,8p,d4,p,8p,d#4,p,8p,f#4`
ESC 4
`Melody:b=150,o=5,d=16:d#3,p,g4,p,d#3,p,a4,p,d#3,p,g4,p,d#3,p,f#4,p,d#3,p,g4,p,d#3,p,a4,p,d#3,p,g4,p,d#3,p,f#4,p,d#3,p,g4,p,d#3,p,a4,p,d#3,p,g4,p,d#3,p,f#4,p,d#3,p,g4,p,d#3,p,a4`

stylesuxx, 2023-04-23:
@GKLove - Could you please also add the artist/game where the tune is from? This allows me to categorize it properly, I am having trouble finding the sources for some of your submissions.

GKLove, 2023-05-30:
Senbonzakura

ESC 1
`Melody:d=16,o=5,b=156:b5,p,c#6,p,f#5,e5,f#5,e5,b5,p,c#6,p,f#5,e5,f#5,e5,b5,p,c#6,p,f#5,e5,f#5,e5,a5,p,g#5,32a5,32g#5,32f#5,32p,p,e5,p,1p,1p`
ESC 2
`Melody:d=16,o=5,b=156:1p,1p,b5,p,c#6,p,f#5,e5,f#5,e5,b5,p,c#6,p,f#5,e5,f#5,e5,b5,p,c#6,p,e6,p,a6,p,g#6,a6,g#6,f#6,e6,p,c#6`
ESC 3
`Melody:d=16,o=5,b=156:b5,p,c#6,p,f#5,e5,f#5,e5,b5,p,c#6,p,f#5,e5,f#5,e5,b5,p,c#6,p,f#5,e5,f#5,e5,a5,p,g#5,32a5,32g#5,32f#5,32p,p,e5,p,1p,1p`
ESC 4
`Melody:d=16,o=5,b=156:1p,1p,b5,p,c#6,p,f#5,e5,f#5,e5,b5,p,c#6,p,f#5,e5,f#5,e5,b5,p,c#6,p,e6,p,a6,p,g#6,a6,g#6,f#6,e6,p,c#6`

GKLove, 2023-05-30:
Lemon-米津玄師

ESC 1
`Melody:b=86,o=5,d=16:b,c#6,d#6,p,b,g#,8p,c#6,p,a#,p,f#,d#,8p,a#,p,g#,p,f#,b4,8p,f#,p,d#,4p,p,c#5,d#5,e5,8p,p,b5,p,a#5,b5,f#5,8p,p,e5,p,d#5,e5,f5,8p,p,b5,p,a#5,g#5,g5`
ESC 2
`Melody:b=86,o=5,d=16:8p,g#4,4p,8p,p,f#4,4p,8p,p,e4,4p,8p,p,d#4,4p,8p,p,e4,4p,8p,p,d#4,4p,8p,p,d4,4p,8p,p,d#4`
ESC 3
`Melody:b=86,o=5,d=16:8p,d#4,4p,8p,p,c#4,4p,8p,p,g#3,4p,8p,p,b3,4p,8p,p,g#3,4p,8p,p,f#3,4p,8p,p,f3,4p,8p,p,g3`
ESC 4
`Melody:b=86,o=5,d=16:8p,g#3,4p,8p,p,f#3,4p,8p,p,e3,4p,8p,p,d#3,4p,8p,p,e3,4p,8p,p,d#3,4p,8p,p,d#3,4p,8p,p,d#3`

GKLove, 2023-05-30:
强风オールバック

ESC 1
`Melody:o=5,b=135,d=8:d5,16p,32g4,32p,g4,16p,32f4,32p,f4,16g4,32g4,32p,g4,p,g4,a#4,p,d5,c5,1p,1p,1p,1p,1p,1p`
ESC 2
`Melody:o=5,b=135,d=8:1p,1p,g4,16p,32g4,32p,g4,f4,g4,p,g4,f4,g4,a#4,a4,f4,d4,1p,1p,1p,1p`
ESC 3
`Melody:o=5,b=135,d=8:1p,1p,1p,2p,4p,g4,p,d5,p,g4,16f4,32f4,32p,f4,16g4,32g4,32p,g4,p,g4,16a#4,32a#4,32p,a#4,c5,d5,1p,1p`
ESC 4
`Melody:o=5,b=135,d=8:1p,1p,1p,1p,1p,1p, g4,16g4,16f4,g4,16g4,16f4,g4,a#4,c5,d5,c5,a#4,g4,f4,g4`

padiyul, 2023-07-22:
Star Trek Next Generation Polyphonic
ESC 1
`Melody:b=160,o=5,d=4:2f,8p,8d,2d#,8c,8g,f.,d.,p,8d#,8f,8g,8d#,f.,d#.,d.,8d,8p,8d,2f.,1p`
ESC 2
`Melody:b=160,o=4,d=4:2a#,8a#,8p,2g#,p,a#.,p.,8p,8g#,2p,f.,g.,f.,g.,2a.,1p`
ESC 3
`Melody:b=160,o=5,d=4:2f.,2c.,2p.,2d#,p,d.,d#.,d,8p,d,8p,2c.,1p`
ESC 4
`Melody:b=160,o=3,d=4:16a#,16p,16a#,16p,16a#,16p,a#.,16g#,16p,16g#,16p,16g#,16p,g#.,16a#,16p,16a#,16p,16a#,16p,8a#,8f,8a#,g#,2p,2a#.,16c4,16p,8c4,a#.,p.,8p,16c4,16p,8c4,f.,2p.`

LegoRJ35, 2023-08-10:
My little Pony theme1
All ESCs
MyLittlePony:d=8,o=7,b=126:f#5,a5,f#5,4e5,4a.5,d5,e5,d5,4c#5,4a.4

LegoRJ35, 2023-08-10:
My little Pony theme2
ESC1:
MyLittlePony:d=8,o=7,b=126:f#5,a5,f#5,4e5,4a.5,d5,e5,d5,4c#5,4a.4,4b4,4c#5,4d5,4e5,2d.5
ESC2:
MyLittlePony:d=8,o=7,b=126:d3,d4,a4,f#4,e3,e4,a4,a3,b3,f#4,b4,f#4,a3,e4,a4,a3,4g3,4a3,4b3,4c#4,2d.4
ESC3:
MyLittlePony:d=8,o=7,b=126:1p,1p,4d5,4e5,4f#5,4g5,2f#.5
ESC4:
MyLittlePony:d=8,o=7,b=126:f#6,a6,f#6,4e6,4a.6,d6,e6,d6,4c#6,4a.5,4g5,4a5,4b5,4c#6,2a.5

LegoRJ35, 2023-08-10:
K-391 - The is Felicitas
ESC1:
k391:b=126,o=7,d=8:p,9f5,9g5,9f5,9f5,9e5,4f5,4f5,4f5,4e5,4e5,7p,9f5,9g5,9f5,9f5,9e5,4f5,7p,16f5,7p,32a#5,64p,16a#5,64p,32a#5,64p,4g5,4e5
ESC2:
k391:b=126,o=7,d=8:p,9d4,9d4,9d4,4a#4,4a#4,4g4,4g4,4c5,4c5,7p,9d4,9d4,9d4,4a#4,4a#4,7p,16g4,7p,32g4,64p,16g4,64p,32g4,64p,4c5,4c5
ESC3:
k391:b=126,o=7,d=8:p,9d4,9d4,9d4,4a#3,4a#3,4g3,4g3,4c4,4c4,7p,9d4,9d4,9d4,4a#3,4a#3,7p,16g3,7p,32g3,64p,16g3,64p,32g3,64p,4c4,4c4
ESC4:
k391:b=126,o=7,d=8:p,9f5,9f5,9f5,4f5,4f5,4f5,4f5,4e5,4e5,7p,9f5,9f5,9f5,4f5,4f5,7p,16f5,7p,32a#5,64p,16a#5,64p,32a#5,64p,4e5,4e5

olegshevtsov, 2023-11-29:
Name of the song: Гімн України
All ESCs
UAAnthem:b=150,o=5,d=4:e,8e,8e,8d,8e,8f,g,8f,e,d,c,e,b4,e,a4,a4

FFFFH, 2023-12-25:
Peppa Pig

All ESCs
PeppaPig:b=300,o=5,d=4:g,p,e,c,d,p,g4,p,g4,b4,d,f,e,p,c,p

btastic, 2023-12-31:
T-Motor

All ESCs
tmotor:b=130,o=5,d=8:c,16f,16e,f,e,4c

GKLove, 2024-04-02:
Vampire DECO*27

ESC 1
`bluejay:b=163,o=4,d=8:a5,g5,f5,g5,p,4f5,f6,p,a5,16a5,g5,16p,f5,p,f6,p,1p,1p,1p,1p,1p,1p`

ESC 2
`bluejay:b=163,o=4,d=8:1p,1p,1p,1p,4p,a5,d6,e6,4f6,a5,g5,p,f6,p,g6,16g6,32g6,32p,g6,f6,g6,4a6,1p,2p,p`

ESC 3
`bluejay:b=163,o=4,d=8:1p,1p,a5,g5,16g5,16f5,p,g5,p,f5,g5,32f5,32p,16f5,a5,16p,16g5,p,d6,16d6,32d6,32p,d6,c#6,d6,e6,1p,1p,1p,2p,4p`

ESC 4
`bluejay:b=163,o=4,d=8:1p,1p,1p,1p,1p,1p,4p,p,g6,16f6,32f6,32p,f6,p,4a6,g6,f6,g6,2f6`

krkvsk, 2024-06-14:
Poly version of  "ой у лузі червона калина" 

kalyna:d=4,o=5,b=260: 4d6,4p, 4d6, 4p, 2f6, 4p, 4d6,  4e6, 4f6, 4e6, 4d6,4c#6, 4p, 4a5,4p, 2d6,4p, 4e6,4f6, 4p, 4e6, 4p, 4d6

kalyna:d=4,o=5,b=260: 4d4, 4f4,4d4, 4f4,4d4, 4f4,4d4, 4f4, 4e4, 4g4,4e4, 4g4, 4a4,4c#,4a4,4c#, 4d4, 4f4,4d4, 4f4,  
4c4, 4e4,4c4, 4e4,4d4, 4f4,4d4, 4f4

Mask off

mask_off:d=4,o=5,b=200: 2e, 8g, 8b ,8g, 1a, 8b,8a,8g,4f#, 4e, 8d, 8b4, 1e
mask_off:d=4,o=5,b=200: 1e2, 8p, 1a2, 8p, 1d2, 32p, 1e2
mask_off:d=4,o=5,b=200: 1e3, 8p, 1a3, 8p, 1d3, 32p, 1e3
mask_off:d=4,o=5,b=200: 2e6, 8g6, 8b6 ,8g6, 1a6, 8b6,8a6,8g6,4f#6, 4e6, 8d6, 8b, 1e6

Rodion49, 2024-07-25:
as_Wollen_Wir_Trinken:o=6,d=4,b=126:16d,16p,8d,16c,16b5,c,a5.,16d,16p,8d,8c,8b5,16d,16p,8d,16c,16b5,4c,a5,8b5,8g5,4a5

krkvsk, 2024-07-25:
"police academy"  blue oyster bar

Esc1
for_sofin:b=125,o=5,d=4:8e5,8c6,8b5,8a5,8e5,1c5,p, 8p,8e5,8c6,8b5,8a5,8e5,g5,8p,8f5, 64p,2f5
Esc2
for_sofin:b=125,o=5,d=4:2p,8p,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8d4,8e3,8d4,8e3,8d4,8e3,8d4,8e3,8d4
Esc3
for_sofin:b=125,o=5,d=4:8e5,8c6,8b5,8a5,8e5,1c5,p, 8p,8e5,8c6,8b5,8a5,8e5,g5,8p,8f5, 64p,2f5
Esc4
for_sofin:b=125,o=5,d=4:2p,8p,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8d4,8e3,8d4,8e3,8d4,8e3,8d4,8e3,8d4

RainityZhou, 2024-08-06:
ESC1：
Melody:b=100,o=5,d=8:4d#6,d#6,4c#6,b,4a#,4a#,d#,f,f#,g#,a#,g#,f#,d#,4f#,f#,4f,f,p,4d#6,4c#6,b,4a#,4a#,d#,f,f#,g#,a#,g#,f#,f,d#

ESC2：
Melody:b=100,o=4,d=8:
d#,16d#5,16p,d#5,c#,16c#5,16p,c#5,d#,16d#5,16p,d#5,d#,16d#5,16p,d#5,
b,16b5,16p,b5,b,16b5,16p,b5,g#,16g#5,16p,g#5,a#,16a#5,16p,a#5,
p,d#,16p,16d#,c#,16c#5,16p,c#5,d#,16d#5,16p,d#5,d#,16d#5,16p,d#5,
b,16b5,16p,b5,c#,16c#5,16p,c#5,d#5

ESC3：
Melody:b=100,o=4,d=8:
p,16a#,16p,a#,p,16g#,16p,g#,p,16a#,16p,a#,p,16a#,16p,a#,
p,16f#,16p,f#,p,16f#,16p,f#,p,16d#,16p,d#,p,16f,16p,f,
4p,p,4c#5,d5,4d#5,d#5,4a#5,a#5,
4g#5,g#5,4c#5,c#5

ESC4：
Melody:b=100,o=4,d=8:
4a#5,a#5,4g#5,f#5,2a#5,f#5,g#5,
2f#5,4f#5,4g#5,g#5,4a#5,a#5,
p,a#,16p,16a#,p,16g#,16p,g#,p,16a#,16p,a#,p,16a#,16p,a#,
p,16f#,16p,f#,p,16g#,16p,g#,a#

3vilson, 2024-09-21:
Kim Possible Pager

pager:d=16,o=5,b=200:d6,8p,d6,8p,f6,32p,d6

WaspFPV, 2024-10-11:
The X-Files Theme (Polyphonic)
Not my work. Source: https://www.youtube.com/watch?v=OP0oCVohopc

ESC1:
Xfiles1:b=384,o=5,d=4:a4,c,e,f,c,e,f,c,e,f,c,e,a,2p,a,2p,a

ESC2:
Xfiles2:b=384,o=5,d=4:a4,c,e,f,c,e,f,1p,1p,1p

ESC3:
Xfiles3:b=384,o=5,d=4:a4,c,e,f,1p,1p,1p,2p,p

ESC4:
Xfiles4:b=384,o=4,d=4:a,e,a,e,a,e,a,e,a,e,a,e,a,e,a,e,a,e,a

RIM0T0, 2025-05-01:
Angry Birds theme (short)
--------------------------------------
ESC1: AngryBirdsTreble: d=4,o=5,b=250: 8e, 8f#, g, e, b, 8e, 8f#, g, 16b, 16b, 16b, 16p, b, p, 8b6, 8c7, 8b6, 8a6, 16g6, 16g6, 16g6, 16p, 8g6, 8f#6, 2e6

ESC2: AngryBirdsBass: d=4,o=3,b=250: p, e, e, e, 8e, 32e, 32e, 32e, 32p, 1e, 2g, 2f#, 2e

ESC3: AngryBirdsBass: d=4,o=3,b=250: p, e, e, e, 8e, 32e, 32e, 32e, 32p, 1e, 2g, 2f#, 2e

ESC4: AngryBirdsTreble: d=4,o=5,b=250: 8e, 8f#, g, e, b, 8e, 8f#, g, 16b, 16b, 16b, 16p, b, p, 8b6, 8c7, 8b6, 8a6, 16g6, 16g6, 16g6, 16p, 8g6, 8f#6, 2e6

syao99, 2025-06-01:
Doom theme in the original key. Based on https://www.youtube.com/watch?v=ME3ZA7A7HIQ

doom:b=420,o=4,d=32:8g4,8e4,8b3,8e4,8g4,8e4,8g4,8b4,8g4,8e4,8g4,8e4,8g4,8b4,8e5,8g5,8e3,8p,8e3,8p,4e,8e3,8p,8e3,8p,4d,8e3,8p,8e3,8p,4c,8e3,8p,8e3,8p,4a#3,8e3,8p,8e3,8p,4b3,4c,8e3,8p,8e3,8p,4e,8e3,8p,8e3,8p,4d,8e3,8p,8e3,8p,4c,8e3,8p,8e3,8p,1a#3

krkvsk, 2025-09-28:
Smaragdove nebo poly 
ESC 1
nebo:b=130,o=4,d=4:e,8b,8p,8b,8a,8b,e,8b,8p, 8b,8p, 8a,b,d5,8c5,8p,8c5,8b, 8c5,e,8c5,8p,8c5,8p, 8b,c5,d5

ESC 2
nebo:b=130,o=4,d=4:4p, 8e3,8b3,8e3,8b3,8e3,8b3,8e3,8b3, 8e3,8b3,8e3,8b3,8e3,8b3,8e3,8b3,
 8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3

ESC 3
nebo:b=130,o=4,d=4:4p, 8e3,8b3,8e3,8b3,8e3,8b3,8e3,8b3, 8e3,8b3,8e3,8b3,8e3,8b3,8e3,8b3,
8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3,8a3,8e3

ESC 4
Melody:b=130,o=4,d=4:e,8b,8p,8b,8a,8b,e,8b,8p, 8b,8p, 8a,b,d5,8c5,8p,8c5,8b, 8c5,e,8c5,8p,8c5,8p, 8b,c5,d5

krkvsk, 2025-09-28:
Legendary theme of bar "blue ouster" from Police academy movie

Esc1
sofin_pidr:b=125,o=5,d=4:8e5,8c6,8b5,8a5,8e5,1c5,p, 8p,8e5,8c6,8b5,8a5,8e5,g5,8p,8f5, 64p,2f5
Esc2
sofin_pidr:b=125,o=5,d=4:2p,8p,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8d4,8e3,8d4,8e3,8d4,8e3,8d4,8e3,8d4
Esc3
sofin_pidr:b=125,o=5,d=4:8e5,8c6,8b5,8a5,8e5,1c5,p, 8p,8e5,8c6,8b5,8a5,8e5,g5,8p,8f5, 64p,2f5
Esc4
sofin_pidr:b=125,o=5,d=4:2p,8p,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8c4,8e3,8d4,8e3,8d4,8e3,8d4,8e3,8d4,8e3,8d4

Blunaa, 2025-11-25:
Kenji Kawai - Making of Cyborg (Ghost in the Shell Theme):

s:d=4,o=5,b=300:2p,16a4,1a#4,1c,2p,d,d#,1f,2p,32p,32f,2f#,1f,1p,1p,2p,2d,1d#,1d#,1p,d#,d,c,a#4,c,2p,1p,1p,2p,2a#4,1c,2p,d,d#,1f,2p,32p,32f,2f#,1f,1p,1p,2p,2d,1d#,1d#,1p,d#,d,2c,16p,1a#4,2p,8p,4c,4c,1c

Might i ask which tools you used? i would like to make a second version with more channels. Is there a kinda multi-channel composer tool which i could use?

thx for the nice work. 73.