stylesuxx/esc-configurator #96: [Fix request]Move the bottom status bar information slightly to the center

DusKing1, 2021-04-10:
in some new phone like my oneplus8 as we have a round corner, the version info and port info will be cut a little. As shown:

[image]

stylesuxx, 2021-04-10:
Do you know how to remote debug with chrome? It would be cool if you could  tell me how much I need to offset it from the sides.

DusKing1, 2021-04-10:
ahh I don't know.., but in my case, "Port blabla" need to be moved right about 2 chars and so did the version string

stylesuxx, 2021-04-10:
I am sure there is also not enough space to display the battery information. Generally it might be a better idea to move all of this into the center and shorten the labels on mobile.

Generally I think mobile needs some adaptations.

DusKing1, 2021-04-10:
So you decide to move put the battery information also at the bottom bar? I didn't keep a good track of your latest development.

stylesuxx, 2021-04-10:
Yes.
[image]

But I actually might move it into the Motor control box, since I can only read it before connecting to the ESCs anyway.

closed 2021-04-10 by stylesuxx