stylesuxx/esc-configurator #98: Flashing error

mathiasvr, 2021-04-10:
When the firmware file has not been loaded from web, there is time to press the "Flash" button multiple times potentially leading to corruption.
I'm not sure what caused the problem since this also caused strange behavior when disconnecting/connecting and i ultimately had to refresh.

stylesuxx, 2021-04-10:
By "not loaded from web" you mean flashing a local file? Or do you mean a file that has been loaded from web previously and the cached version was used?

mathiasvr, 2021-04-10:
W the file has not been cached it takes some time to download it.

stylesuxx, 2021-04-10:
Yes, good catch - had to throttle my internet to be able to reproduce. Still, it will "only" flash the firmware multiple times. Interesting that you ended up with a broken state.

mathiasvr, 2021-04-10:
Hm, broken/flash failed only seems to happen when i spam the button a lot.

commit b4e23048 (2021-04-10): Change to flashing state as soon as firmware is being donwloaded. Resolves #98
closed 2021-04-20 by stylesuxx