stylesuxx/esc-configurator #103: Do not migrates settings between different firmwares

stylesuxx, 2021-04-18:
As discussed with @mathiasvr it should no longer be possible to migrate settings between different firmware versions - this might lead to errors since settings that are named the same way do not necessarily act the same way in every firmware.

This is a bit of a loss of convenience, since - for example - Motor directions need to be set after changing firmware, but we gain confidence not to potentially break gear.

It is probably a good idea to add a warning in the flash window that changing firmware will result in the defaults being flashed.

On the other hand, there is a checkbox during the flash process that allows for migrations to be disabled - Maybe this should be re-worded so that activating it will attempt migration between firmware, but the default is not to migrate between different firmware.

mathiasvr, 2021-04-18:
I got sidetracked when looking into this because I ended up trying something a bit too complicated. IIRC it should be safe to migrate settings between all blheli_s based firmware except bluejay so maybe it should be allowed for convenience?

Not sure if it was the right decision but bluejay will probably further complicate integration with the configurator as it's on a path of becoming even less compatible with blheli_s.

stylesuxx, 2021-04-18:
> IIRC it should be safe to migrate settings between all blheli_s based firmware except bluejay so maybe it should be allowed for convenience?

Well, that would then just be BLHeli_S. I was under the impression that you just want to remove migrations between different firmware - so no migration from BLHlei_S to Bluejay and vice versa. Maybe I understood you wrongly - should migrations between different Bluejay versions also no longer be possible?

mathiasvr, 2021-04-18:
I mean migrations between BLHeli_S/JESC/BLHeli_M should be fine maybe except from PPM settings.
Migration between individual versions of Bluejay is okay, though downgrading could pose some problems. I think that might be fine in the beta phase, but we could also disallow migration to a lower version. This only applies to bluejay so maybe not worth it.

TLDR: The only significant problem is migrating settings between Bluejay and BLHeli_S/JESC/BLHeli_M.

stylesuxx, 2021-04-18:
OK, now I get it.

Not allowing migrations during downgrade is not a big deal. this could be a setting in the source files. Also cross migrations could be disallowed in the source files.

commit 6056ee5a (2021-04-24): Do not migrate settings between different firmwares by default. Allow users to force migration between firmwares. Inform user that his settings might get lost. Resolves #103