stylesuxx/esc-configurator #104: COM port selector

JesusOfOz, 2021-04-21:
Hi there,
I am on a WIN10 laptop in an Opera browser and though it sees the STMicroelectronics port and I select it, nothing happens - tried selecting other ports and the same thing happen, they highlight in the chooser window and that's it - is there a secret key combo to kick it in?
Gruss aus Dänemark.
Frank

stylesuxx, 2021-04-21:
This functionality is not supported by Opera, at least according to https://caniuse.com/mdn-api_serialport - but the same settings in regards to permissions for serial and USB devices are available as in stock Chromium.

I'd say that this is a bug in Opera, since all the web Serial enabled pages behave the same way eg.: https://unjavascripter-web-serial-example.glitch.me

If you don't want to use Chrome directly, I suggest using Vivaldi instead.

JesusOfOz, 2021-04-21:
Thanks for the quick and thorough reply :-)

I had a hunch we were down that road, just thought I'd ask before I acted on it.

I have been using Mathias's configurator for a while, however for "field tinkering" it is great with a web interface that may be used with OTG on a tablet or smartphone so I can leave the laptop behind at home.

Keep it up, thanks again.

Frank

closed 2021-04-21 by JesusOfOz

stylesuxx, 2021-04-21:
I opened a bug report with Opera, let's see what they answer. Regarding tablet or smartphone, please check in advance. Not all android phones support webUsb properly, due to a Chrome for Android "feature" preventing Chrome to claim USB devices in certain cases, see here => #56