stylesuxx/esc-configurator #107: Import/Export settings

stylesuxx, 2021-04-27:
A user should be able to export his current ESC settings and import them back in.

The export should consist of the used firmware and version. Import of the settings should only be possible if those criteria match and an equal amount of ESC's is present.

The export should be a JSON object.

cinderblock, 2022-05-02:
I would vote for the "equal amount of ESC's present" feature to be optional. Sometimes I want to copy from one ESC to another and they many not be on the same channel/position.

stylesuxx, 2022-05-02:
> I would vote for the "equal amount of ESC's present" feature to be optional. Sometimes I want to copy from one ESC to another and they many not be on the same channel/position.

I guess this would then actually be two different functions:
1. Export/Import - based on firmware, version and count
2. Copy settings from other ESC - Each ESC would have a drop down from which you would select the ESC you want to copy settings from - this would only make sense for BLHeli_S though, since individual ESC's only have one setting on Bluejay.

Is this what you'd be looking for?

cinderblock, 2022-05-02:
Hmmm, I'm not following entirely.

The (very) specific use case I'm (particularly) interested in is:
1. "Connect" to FC
2. Plug in ESC to FC
3. Flash known good settings (and firmware??) to the ESC
4. Repeat 2-3 for a small number of ESCs (not enough to justify using a C2 programming interface)

Of course, I'm not expecting a feature that only I want. It just feels like a standardized "block" of settings for one (or many) ESC(s) would be a valuable packing feature.

I'm a special use case because I'm working on a project that only uses one ESC, so I'm not sure what number of buttons should be added - I've only really ever looked at the interface with a single ESC!

So maybe on each ESC's configuration a pair of buttons labeled "Save/Load ESC configuration". This would be different from a pair of buttons labeled "Save/Load all ESC configurations" on the main footer bar that only does all controllers at once in the matched configuration as you mentioned.

Maybe a different file extension for each type of json file, like "`*.escs.json`" and "`*.esc.json`" or "`*.blheli.json`" and "`*.bj.json`", would help keep the disparate types of saved configurations separate.

stylesuxx, 2022-12-12:
The firmware can now be dumped as a hex file . This allows for full backup of settings for a specific ESC (by Layout).

closed 2022-12-12 by stylesuxx