stylesuxx/esc-configurator #108: AM32 melodies

stylesuxx, 2021-04-28:
@AlkaMotors now also implemented startup melodies for AM32 and it takes the same format that bluejay is currently using.

EEPROM layout need to be updated though. @AlkaMotors can you please update the EEPROM Layout here => https://github.com/stylesuxx/esc-configurator/blob/develop/src/sources/AM32/eeprom.js Would be cool if we could keep the same [name as with bluejay](https://github.com/stylesuxx/esc-configurator/blob/6056ee5aaf1176a1918840b3d9b7c95a49c9438a/src/sources/Bluejay/eeprom.js#L1313), then no other changes should be needed.

AlkaMotors, 2021-04-29:
I haven't rolled it into the firmware yet. I am still trying to finalize things. The idea is as long as there is a valid bluejay array of 128bytes after the 48byte eeprom.. it will play it. I'll take a look at your repo at the bluejay format and keep everything the same in terms of naming and eeprom layout.

stylesuxx, 2021-07-13:
This should already be possible with the latest firmware version.

* [ ] Check that it is
* [ ] Enable melody editor for the latest firmware
* [ ] Expand testing protocol

stylesuxx, 2021-10-29:
Everything is looking good with AM32 v1.78

stylesuxx, 2021-11-14:
Deployed with v0.20.0

closed 2021-11-14 by stylesuxx