stylesuxx/esc-configurator #144: Cache firmware versions for offline use

mathiasvr, 2021-09-10:
It would be nice to be able to use the app offline and view/flash previous versions like blheli-configurator.

stylesuxx, 2021-09-10:
Yeah - you can actually do that right now. Firmware is cached the first time you download it. Every time after that you flash it from local storage. This allows ist to also be used offline with the firmware that has already been downloaded before.

But I assume something is not working as expected,?

mathiasvr, 2021-09-10:
Ah cool, I think the list of versions is not cached then?

stylesuxx, 2021-09-11:
Ah, I think I understand your issue. No - if the app is offline it uses the "built in" list of versions, it will still show you everything, not just the stuff that has been cached. Files are cached with the URL to the file as a key, so it's a bit difficult to only show the options for the files that are cached, if that is what you are looking for?

mathiasvr, 2021-09-11:
If I open the PWA or website without internet I can only see e.g. version 11 and lower of bluejay, but I have flashed newer versions before when online.

stylesuxx, 2021-09-11:
You are right - the 'local' file is outdated => https://github.com/stylesuxx/esc-configurator/blob/master/src/sources/Bluejay/versions.json Would you mind commiting the latest one?

mathiasvr, 2021-09-11:
I blheli configurator this file is cached, I think this would make sense for ESC configurator as well since it fetches it online. This way the app doesn't need to be updated for new versions. If you would rather have the versions built in and update them with the app, it should never fetch them online imo.

stylesuxx, 2021-09-11:
Hmm, yes - good point with caching. I might look into refactoring that part then.
The app still needs to be updated for new versions of AM32, but yeah I get your point.