stylesuxx/esc-configurator #150: AM32 stopped reading ESC after saving settings.

howels, 2021-10-01:
Was able to use configurator earlier today, but after saving settings on an AM32 ESC the ESC is no longer detected. I hear 3 ascending tones when clicking the "Read settings" button but no settings page is shown. 

FC firmware is iNav, tried 3.0.1 and then upgraded to 3.0.2 to see if it would make a difference but no change.

Log contents is this:
```
2021/10/01 22:48:26: Plugged in
2021/10/01 22:48:53: Opened serial port
2021/10/01 22:48:53: MultiWii API version received: 2.4.0
2021/10/01 22:48:53: Flight controller info, identifier: INAV version: 3.0.2
2021/10/01 22:48:53: Running firmware released on: Sep 15 2021 10:08:18
2021/10/01 22:48:53: Board: MF4S, version: 0
2021/10/01 22:48:53: Unique device ID received - 0x043004c594b501320313457
2021/10/01 22:48:54: FC reports unsupported message error: 130
2021/10/01 22:48:55: FC reports unsupported message error: 130
2021/10/01 22:48:56: FC reports unsupported message error: 130
2021/10/01 22:48:57: FC reports unsupported message error: 130
2021/10/01 22:48:58: FC reports unsupported message error: 130
2021/10/01 22:48:59: FC reports unsupported message error: 130
2021/10/01 22:49:00: FC reports unsupported message error: 130
2021/10/01 22:49:01: FC reports unsupported message error: 130
2021/10/01 22:49:02: FC reports unsupported message error: 130
2021/10/01 22:49:03: FC reports unsupported message error: 130
2021/10/01 22:49:05: Trying to read 1 ESC's
2021/10/01 22:49:05: TX: cmd_DeviceInitFlash - 0x0 
2021/10/01 22:49:05: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2021/10/01 22:49:05: Failed reading ESC 1
2021/10/01 22:49:05: Error: Message not OK 
2021/10/01 22:49:23: Trying to read 1 ESC's
2021/10/01 22:49:23: TX: cmd_DeviceInitFlash - 0x0 
2021/10/01 22:49:23: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2021/10/01 22:49:23: Failed reading ESC 1
2021/10/01 22:49:23: Error: Message not OK 
2021/10/01 22:49:23: Trying to read 1 ESC's
2021/10/01 22:49:23: TX: cmd_DeviceInitFlash - 0x0 
2021/10/01 22:49:23: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2021/10/01 22:49:23: Failed reading ESC 1
2021/10/01 22:49:23: Error: Message not OK 
2021/10/01 22:49:26: Trying to read 1 ESC's
2021/10/01 22:49:26: TX: cmd_DeviceInitFlash - 0x0 
2021/10/01 22:49:26: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2021/10/01 22:49:26: Failed reading ESC 1
2021/10/01 22:49:26: Error: Message not OK 
2021/10/01 22:49:26: Trying to read 1 ESC's
2021/10/01 22:49:26: TX: cmd_DeviceInitFlash - 0x0 
2021/10/01 22:49:26: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2021/10/01 22:49:26: Failed reading ESC 1
2021/10/01 22:49:26: Error: Message not OK 
```

stylesuxx, 2021-10-01:
Could you please post the complete debug log - bottom left you find the button to download it. And this happened directly after saving changes? Could you also please tell me which version of AM32 you are running (including bootloader version).

howels, 2021-10-02:
Latest AM32 as of today - release 1.74 `IFLIGHT_1.74.hex`, bootloader `PB4_BOOTLOADER_V10.bin`, EEPROM version `eeprom_version_1_7c00.bin`

The full debug log shows entries back to April.... So I'm just attaching today's entries:

```
2021/10/02 12:21:57: Plugged in
2021/10/02 12:22:43: Plugged in
2021/10/02 12:22:46: Opened serial port
2021/10/02 12:22:46: MultiWii API version received: 2.4.0
2021/10/02 12:22:46: Flight controller info, identifier: INAV version: 3.0.2
2021/10/02 12:22:46: Running firmware released on: Sep 15 2021 10:08:18
2021/10/02 12:22:46: Board: MF4S, version: 0
2021/10/02 12:22:46: Unique device ID received - 0x043004c594b501320313457
2021/10/02 12:22:47: FC reports unsupported message error: 130
2021/10/02 12:22:48: FC reports unsupported message error: 130
2021/10/02 12:22:49: FC reports unsupported message error: 130
2021/10/02 12:22:50: FC reports unsupported message error: 130
2021/10/02 12:22:51: FC reports unsupported message error: 130
2021/10/02 12:22:52: FC reports unsupported message error: 130
2021/10/02 12:22:53: FC reports unsupported message error: 130
2021/10/02 12:22:54: FC reports unsupported message error: 130
2021/10/02 12:22:55: FC reports unsupported message error: 130
2021/10/02 12:22:56: FC reports unsupported message error: 130
2021/10/02 12:22:57: FC reports unsupported message error: 130
2021/10/02 12:22:58: FC reports unsupported message error: 130
2021/10/02 12:22:59: FC reports unsupported message error: 130
2021/10/02 12:23:00: FC reports unsupported message error: 130
2021/10/02 12:23:02: Trying to read 1 ESC's
2021/10/02 12:23:02: TX: cmd_DeviceInitFlash - 0x0 
2021/10/02 12:23:02: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2021/10/02 12:23:02: Failed reading ESC 1
2021/10/02 12:23:02: Error: Message not OK 
```

stylesuxx, 2021-10-03:
Very interesting.

How long are you wainting before reading the settings? Is there by any chance a melody playing? If so, please wait before trying to read the settings.

Can you read it with the stand alone AM32 configurator?

Which setting did you change before it stopped working?

The unsupported message error is also very interesting - could you do me a favor and check if it is behaving the same way with Betaflight flashed?

howels, 2021-10-03:
> Very interesting.
>
> How long are you wainting before reading the settings? Is there by any chance a melody playing? If so, please wait before trying to read the settings.
>
> Can you read it with the [...]

Waiting at least 10 seconds.  There is no melody playing, I disabled beeps with a `beeper -ON_USB` in iNav. I don't have a version of the standalone configurator for my OS (linux). I changed the motor KV and the startup power settings, those may be linked to it. I did a "reset to defaults" then changed motor speed and startup power.

I can't flash betaflight to this I'm afraid, I think I'll take the thing apart and do a clean ST-Link flash when I next need to change something. Seems that I've got a bad flash on here but now it's buried inside an RC car.

Thanks for the suggestions!

stylesuxx, 2021-10-03:
OK, please keep me posted. I will try if I can reproduce it, but so far had no luck. I need to flash iNav and see if that might be the issue...

There actually is a configurator for Linux too -> https://drive.google.com/file/d/164mF-7khqH822kLzF8sP5VAckHecoKGG/view the link is from the Readme of the AM32 repository. I am on Linux myself and have used this initially to validate the configurators functionality against.

stylesuxx, 2022-08-09:
Hey @howels any news on this? If not I would close the issue and blame a "bad" flash ;-)

howels, 2022-08-09:
Yeah, I think it was too long a ESC signal wire. I haven't opened the car up since.