stylesuxx/esc-configurator #166: Can not read setup on Kakute F7 mini v2 + Tekko32 F3 mini

PMantix, 2022-01-09:
When trying to "Read Setup" with the Kakute F7 mini v2 stack (https://shop.holybro.com/kakute-f7-mini-stacks_p1194.html) the following error is encountered:

----

2022/01/09 17:11:51: Opened serial port
2022/01/09 17:11:51: MultiWii API version received: 1.43.0
2022/01/09 17:11:51: Flight controller info, identifier: BTFL version: 4.2.11
2022/01/09 17:11:51: Running firmware released on: Nov  9 2021 20:29:04
2022/01/09 17:11:51: Board: S745, version: 0
2022/01/09 17:11:51: Unique device ID received - 0x037004f3039510535383236
2022/01/09 17:12:01: Trying to read 4 ESCs
2022/01/09 17:12:01: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2022/01/09 17:12:01: Failed reading ESC 1
2022/01/09 17:12:01: Error: Message not OK 
2022/01/09 17:12:01: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2022/01/09 17:12:01: Failed reading ESC 2
2022/01/09 17:12:01: Error: Message not OK 
2022/01/09 17:12:01: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2022/01/09 17:12:01: Failed reading ESC 3
2022/01/09 17:12:01: Error: Message not OK 
2022/01/09 17:12:02: Failed processing command cmd_DeviceInitFlash after 0 retries. 
2022/01/09 17:12:02: Failed reading ESC 4
2022/01/09 17:12:02: Error: Message not OK 

----

I've been able to connect other flight controllers to the esc-configurator from this machine, so it might be something specific to this flight controller stack.

Serial port opens OK, and the motors can be spun. Just can not proceed beyond the "Read Setup" button.

stylesuxx, 2022-01-10:
Tekko32 is a BLHeli_32 ESC. The configurator only supports BLHeli_S (and compatible) and AM32.

PMantix, 2022-01-10:
Ah, that explains it! Sorry for the misunderstanding!

closed 2022-01-10 by PMantix