stylesuxx/esc-configurator #167: Motor Direction Won't Flash/Save

ShroomzFPV, 2022-01-11:
When trying to change the direction of the motors, it flashes then reverts back to original direction "normal". Attempted multiple times. Please help if possible, thank you!

[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/7843992/esc-configurator-log.txt)

stylesuxx, 2022-01-11:
I am not sure if we talked on Discord. Anyway, I had someone with the same "issue" there. You have to change the motor direction and then click on "Write" in the bottom right - this will write the changed settings.

"Flash All" or "Flash firmware" will flash the firmware with the default settings.

ShroomzFPV, 2022-01-11:
Wow. I should have picked up on that. Thank you SO much stylesuxx!!!

stylesuxx, 2022-01-11:
No worries - since two people ran into this issue in short time, I should probably clarify that somewhere in the configurator...

closed 2022-01-30 by stylesuxx