stylesuxx/esc-configurator #168: Port already in use by another application - try re-connecting

nR2Cncdm, 2022-01-11:
I'm using Brave browser on Mac Mini M1 as well as Intel based MacBook Pro, and I'm unable to connect to my Beebrain BL V3 drone. The log states "Port already in use by another application - try re-connecting".
[image]

stylesuxx, 2022-01-11:
Brave browser does not support Web Serial unfortunately. They disabled the functionality completely, relevant issue is here => https://github.com/brave/brave-browser/issues/13902

There is nothing I can do on my side. If you don't want to use Google Chrome, you might want to try Vivaldi - also Chromium based and does support the Web serial API. As a last resort you could also try the downloadable app from here https://github.com/mathiasvr/esc-configurator-app

nR2Cncdm, 2022-01-11:
> Brave browser does not support Web Serial unfortunately. They disabled the functionality completely, relevant issue is here => brave/brave-browser#13902

Thanks for the quick reply. I’ll try a different browser.

closed 2022-01-11 by nR2Cncdm