stylesuxx/esc-configurator #178: "Port already in use by another application - try re-connecting"

guilleortas, 2022-01-15:
Hi there, I can't find the "Save Debug Log" button, so I will provide an explanation of the issue instead.
My drivers are completely fine and I can access my ESC settings on the legacy Bluejay configurator standalone app or my Betaflight setting using the regular BF configurator.
The issue is that after allowing access to my COM port (yes, I triple-checked it's the correct one), ESC Configurator just prints a line in the log saying "Port already in use by another application - try re-connecting".
I'm attaching a screen capture so you can see exactly what's going on:
https://filetransfer.io/data-package/9wuZTMDG

Do you have any idea what could be happening?

mathiasvr, 2022-01-15:
Did you install PWA with brave?

Brave browser is not supported so you will need to use chrome/edge or install the standalone version.

DireCat, 2022-01-16:
Confirm the issue with Brave browser (no plugins, shields off). Get the same error "Port already in use ..." on Win10 and doesn't see any COM port on Win7, but works fine with Chromium portable.

guilleortas, 2022-01-16:
@mathiasvr indeed, confirmed to be working flawlessly on edge
Seems like brave just doesn't support PWAs by default

stylesuxx, 2022-01-16:
> @mathiasvr indeed, confirmed to be working flawlessly on edge Seems like brave just doesn't support PWAs by default

The problem is, that Brave removed the functionality for web serial, not that they don't support PWAs...

guilleortas, 2022-01-16:
Ah, that's great to know, thanks!
Nothing wrong with ESC Configurator then 👌🏻

closed 2022-01-16 by guilleortas