stylesuxx/esc-configurator #190: Cannot Flash ESC that Have JFLIGHT Firmware Installed

avella007, 2022-01-23:
Found that both the BlueJ App Configurator and the BlueJ Web based Configurator will not allow you to flash BlueJ ESC on my motors that have JFLIGHT 3.2 flashed.  

I tried to use the BlueJ Web Based Configurator to see if I could flash the Bheli_S first and then the BlueJ, but the configurator messed up my RSCs and I had to go back to FFLIGHT to refash the ESCs.

stylesuxx, 2022-01-23:
We would need some information to be actually able to help you. What does "messed up" mean? Which operating system are you on? Can you post the logs from the configurator? Which ESCs are you using?

Generally there are no issues of flashing over JESC...

stylesuxx, 2022-02-08:
Closing this down due to lack of information, feel free to re-open with the requested information.

closed 2022-02-08 by stylesuxx