stylesuxx/esc-configurator #205: Unable to use PWA on Ubuntu 20.04

Stampede10343, 2022-01-28:
I have tried Chrome, Chromium and Opera browsers and I get the 'Choose Serial Port', which opens up a dialog with no options available.

I am on Ubuntu 20.04. The standalone app works fine luckily, but I would like to be able to use the latest version of the app via the web app.

Stampede10343, 2022-01-28:
I just checked Vivaldi and that appears to be working, but I would prefer if Chromium worked.

stylesuxx, 2022-01-28:
Could be outdated versions of Chromium and Chrome. Which version are you running?

Stampede10343, 2022-01-28:
Chromium 97.0.4692.99

stylesuxx, 2022-01-28:
And Chrome?

Stampede10343, 2022-01-28:
I deleted it after I didn't have any luck with it, but it would have been the latest from the Ubuntu official repositories.

stylesuxx, 2022-01-28:
Just tried with a rather old version of Chromium (89.0.4389.114) and have no issues, will check with a newer version.

stylesuxx, 2022-01-28:
Tried 97.0.4692.99 and have no issues whatsoever...

That you are not seeing any options in the menu is strange though. When you check the sites permission - is 'USB devices' and 'Serial Ports' set to 'Ask'?

Stampede10343, 2022-01-29:
> Tried 97.0.4692.99 and have no issues whatsoever...
> 
> That you are not seeing any options in the menu is strange though. When you check the sites permission - is 'USB devices' and 'Serial Ports' set to 'Ask'?

Yeah, tried Ask and Ask (default). As well as re-installing with no luck. No errors in the console or anything I can see.

stylesuxx, 2022-01-29:
Could you post the output of uname -a please? I will have to set up a system as close as possible to yours to try and see if I can re-create this. Not sure where  to start looking to be honest.

> The standalone app works fine luckily,...

By standalone app you are referencing this one => https://github.com/mathiasvr/esc-configurator-app - right?

Stampede10343, 2022-01-29:
That did it!

Particularly this comment:

> 

The denial on /proc/tty/drivers looks relevant.

Can you try the following?

    - close chromium

    - edit (with sudo) /var/lib/snapd/apparmor/profiles/snap.chromium.chromium, add the following line to the profile:

        @{PROC}/tty/drivers r,

    - reload the apparmor profile with: "sudo apparmor_parser -r /var/lib/snapd/apparmor/profiles/snap.chromium.chromium"

    - run "journalctl -f | grep chromium" in a terminal, start chromium, and try to access your device

    - share the updated journalctl output

  Thanks!

stylesuxx, 2022-01-29:
Awesome, thank you. I will have to document that somewhere...

closed 2022-01-30 by stylesuxx