stylesuxx/esc-configurator #206: Bluejay version '0.16' (Layout 204) is not yet supported.

pfreundl, 2022-01-28:
Got this Error after flashing my jhemcu f405aio (non pro) to 0.16 

thank you

here is the debug log:

2021/10/29 19:16:15: Port selected
2021/10/29 19:16:18: Opened serial port
2021/10/29 19:16:18: MultiWii API version received: 1.43.0
2021/10/29 19:16:18: Flight controller info, identifier: BTFL version: 4.2.0
2021/10/29 19:16:18: Running firmware released on: Jun 14 2020 03:05:26
2021/10/29 19:16:18: Board: S7X2, version: 0
2021/10/29 19:16:18: Unique device ID received - 0x0270037465350152030364e
2021/10/29 19:16:28: Trying to read 4 ESCs
2021/10/29 19:16:28: Read ESC 1: G-H-30 - Bluejay (BETA), 0.14, 48kHz
2021/10/29 19:16:28: Read ESC 2: G-H-30 - Bluejay (BETA), 0.14, 48kHz
2021/10/29 19:16:28: Read ESC 3: G-H-30 - Bluejay (BETA), 0.14, 48kHz
2021/10/29 19:16:29: Read ESC 4: G-H-30 - Bluejay (BETA), 0.14, 48kHz
2021/10/29 19:16:29: Done reading ESCs
2021/10/29 19:16:58: Chosen firmware: https://github.com/mathiasvr/bluejay/releases/download/v0.14/G_H_30_48_v0.14.hex 
2021/10/29 19:16:58: Fetching firmware from repository 
2021/10/29 19:16:59: Saved file to local storage 
2021/10/29 19:16:59: Flashing ESC 1
2021/10/29 19:17:09: Flashed ESC 1 - 10.8s
2021/10/29 19:17:10: Bluejay layout found 
2021/10/29 19:17:10: Migrated setting STARTUP_POWER_MIN 
2021/10/29 19:17:10: Migrated setting DITHERING 
2021/10/29 19:17:10: Migrated setting STARTUP_POWER_MAX 
2021/10/29 19:17:10: Migrated setting RPM_POWER_SLOPE 
2021/10/29 19:17:10: Migrated setting MOTOR_DIRECTION 
2021/10/29 19:17:10: Migrated setting COMMUTATION_TIMING 
2021/10/29 19:17:10: Migrated setting BEEP_STRENGTH 
2021/10/29 19:17:10: Migrated setting BEACON_STRENGTH 
2021/10/29 19:17:10: Migrated setting BEACON_DELAY 
2021/10/29 19:17:10: Migrated setting DEMAG_COMPENSATION 
2021/10/29 19:17:10: Migrated setting TEMPERATURE_PROTECTION 
2021/10/29 19:17:10: Migrated setting BRAKE_ON_STOP 
2021/10/29 19:17:10: Migrated setting STARTUP_MELODY 
2021/10/29 19:17:10: No changes - not updating ESC 1
2021/10/29 19:17:10: Flashing ESC 2
2021/10/29 19:17:21: Flashed ESC 2 - 10.7s
2021/10/29 19:17:21: Bluejay layout found 
2021/10/29 19:17:21: Migrated setting STARTUP_POWER_MIN 
2021/10/29 19:17:21: Migrated setting DITHERING 
2021/10/29 19:17:21: Migrated setting STARTUP_POWER_MAX 
2021/10/29 19:17:21: Migrated setting RPM_POWER_SLOPE 
2021/10/29 19:17:21: Migrated setting MOTOR_DIRECTION 
2021/10/29 19:17:21: Migrated setting COMMUTATION_TIMING 
2021/10/29 19:17:21: Migrated setting BEEP_STRENGTH 
2021/10/29 19:17:21: Migrated setting BEACON_STRENGTH 
2021/10/29 19:17:21: Migrated setting BEACON_DELAY 
2021/10/29 19:17:21: Migrated setting DEMAG_COMPENSATION 
2021/10/29 19:17:21: Migrated setting TEMPERATURE_PROTECTION 
2021/10/29 19:17:21: Migrated setting BRAKE_ON_STOP 
2021/10/29 19:17:21: Migrated setting STARTUP_MELODY 
2021/10/29 19:17:21: No changes - not updating ESC 2
2021/10/29 19:17:21: Flashing ESC 3
2021/10/29 19:17:32: Flashed ESC 3 - 10.7s
2021/10/29 19:17:32: Bluejay layout found 
2021/10/29 19:17:32: Migrated setting STARTUP_POWER_MIN 
2021/10/29 19:17:32: Migrated setting DITHERING 
2021/10/29 19:17:32: Migrated setting STARTUP_POWER_MAX 
2021/10/29 19:17:32: Migrated setting RPM_POWER_SLOPE 
2021/10/29 19:17:32: Migrated setting MOTOR_DIRECTION 
2021/10/29 19:17:32: Migrated setting COMMUTATION_TIMING 
2021/10/29 19:17:32: Migrated setting BEEP_STRENGTH 
2021/10/29 19:17:32: Migrated setting BEACON_STRENGTH 
2021/10/29 19:17:32: Migrated setting BEACON_DELAY 
2021/10/29 19:17:32: Migrated setting DEMAG_COMPENSATION 
2021/10/29 19:17:32: Migrated setting TEMPERATURE_PROTECTION 
[... 230 lines cut ...]
2022/01/28 23:59:12: Flight controller info, identifier: BTFL version: 4.3.0
2022/01/28 23:59:12: Running firmware released on: Jan  9 2022 11:40:24
2022/01/28 23:59:12: Board: S405, version: 0
2022/01/28 23:59:12: Unique device ID received - 0x03400503535510738353732
2022/01/28 23:59:20: Trying to read 4 ESCs
2022/01/28 23:59:20: Read ESC 1: G-H-30 - Bluejay (BETA), 0.14, 48kHz
2022/01/28 23:59:20: Read ESC 2: G-H-30 - Bluejay (BETA), 0.14, 48kHz
2022/01/28 23:59:20: Read ESC 3: G-H-30 - Bluejay (BETA), 0.14, 48kHz
2022/01/28 23:59:21: Read ESC 4: G-H-30 - Bluejay (BETA), 0.14, 48kHz
2022/01/28 23:59:21: Done reading ESCs
2022/01/28 23:59:55: Chosen firmware: https://github.com/mathiasvr/bluejay/releases/download/v0.16/G_H_30_24_v0.16.hex 
2022/01/28 23:59:55: Fetching firmware from repository 
2022/01/28 23:59:56: Saved file to local storage 
2022/01/28 23:59:56: Flashing ESC 1
2022/01/29 00:00:07: Flashed ESC 1 - 11.5s
2022/01/29 00:00:08: Layout revision 204 is not yet supported
2022/01/29 00:00:08: Flashing ESC 2
2022/01/29 00:00:19: Flashed ESC 2 - 11.3s
2022/01/29 00:00:19: Layout revision 204 is not yet supported
2022/01/29 00:00:19: Flashing ESC 3
2022/01/29 00:00:30: Flashed ESC 3 - 11.3s
2022/01/29 00:00:31: Layout revision 204 is not yet supported
2022/01/29 00:00:31: Flashing ESC 4
2022/01/29 00:00:42: Flashed ESC 4 - 11.2s
2022/01/29 00:00:42: Layout revision 204 is not yet supported

stylesuxx, 2022-01-28:
Which version is shown to you on the bottom right of the configurator? Seems like you still have an old version cached... You should try a forced reload - Shift + F5

pfreundl, 2022-01-29:
that might have been the problem, now it works :) thank you! also i just found this great nokia melody (credit to spaghetti_hands on yt) and converted it to RTTTL:
melody:b=240,o=5,d=4:8e6,8d6,f#,g#,8c#6,8b,d,e,8b,8a,c#,e,2a

closed 2022-01-29 by pfreundl