stylesuxx/esc-configurator #210: Open Melody Editor did not work for me

natanrufino, 2022-01-30:
I am not sure if I did something wrong, but the Melody Editor are not starting for me yet.... I click on its button, but nothing happens.

stylesuxx, 2022-01-30:
Could you tell me more? Which firmware and version is currently flashed?

natanrufino, 2022-01-30:
Hi there!
Well, I'm using the Betaflight Configurator 10.8.0, firmware 4.3.0 (Release
Candidate 2). And as to the BluJay, I used it today from
https://esc-configurator.com/ website, so I think it is up to date. The ESC
I was trying to use Melody Editor on, it is called "LanRC" it's a 45A 4n1
ESC. Just for reference, here a link for it: LANRC
<https://pt.aliexpress.com/item/1005001776375757.html?gatewayAdapt=glo2bra&spm=a2g0o.9042311.0.0.6ee3b90a3lltpM>
Thanks!

------------------------------------------------
| Natan Rufino
| natanrufino.com.br
| (83) 9.9967.6732

| Telegram <http://telegram.me/natanrufino>

| YouTube <https://www.youtube.com/user/natanrufino7>

| Instagram <http://instagram.com/natanrufino/>

| Facebook <https://www.facebook.com/natanrufino2>

| Twitter <https://twitter.com/natanrufino>

------------------------------------------------

stylesuxx, 2022-01-30:
Which Browser are you using - please tell me the exact version. Which version of Bluejay is currently flashed to those ESCs?

natanrufino, 2022-01-30:
I used Google Chrome. How do I see the exact version of BlueJay I have
accessed and flashed? Thanks

stylesuxx, 2022-01-30:
When you read the ESCs in the configurator you see them listed on the right side, in each box as a headline you see the layout and version you have currently flashed.

natanrufino, 2022-01-31:
Ahhh... Ok!
You mean those J-H-30, for example? Or those 0.17 numbers? I'll send you a
picture...

natanrufino, 2022-01-31:
Hey! Here's a pic of that you mentioned...
[image: image.png]
Is that it? Do you see the ESC2 is "unkown"? It's a 4n1 ESC, but I don't
know why is like that...
thanks bro!

------------------------------------------------
| Natan Rufino
| natanrufino.com.br
| (83) 9.9967.6732

| Telegram <http://telegram.me/natanrufino>

| YouTube <https://www.youtube.com/user/natanrufino7>

| Instagram <http://instagram.com/natanrufino/>

| Facebook <https://www.facebook.com/natanrufino2>

| Twitter <https://twitter.com/natanrufino>

------------------------------------------------

mathiasvr, 2022-01-31:
This means that ESC2 probably failed flashing and is likely why you cannot use the melody editor.
Since the ESC is layout is now "Unknown", you need to manually flash it with the `J-H-30` layout by checking the "Ignore inappropriate MCU and Layout?" option.

natanrufino, 2022-01-31:
Humm! Awesome! I'll give a try right away... Thanks

natanrufino, 2022-01-31:
> This means that ESC2 probably failed flashing and is likely why you cannot use the melody editor. Since the ESC is layout is now "Unknown", you need to manually flash it with the `J-H-30` layout by checking the "Ignore inappropriate MCU and Layout?" option.

Hey! I have just done that, but it looks like nothing different really happened  What am I missing?

natanrufino, 2022-01-31:
Hey guys! I made a way around the problem I was facing... I just installed BLHeliSuite and flashed the ESCs from there and came back to BlueJay and tried it again and now I can access the "Melody Editor". The only it happened was that the old code "J-H-30" changed for "C-H-40". Is there any problem with that? Thank you guys!!

closed 2022-01-31 by natanrufino