stylesuxx/esc-configurator #223: problem with Mamba F25 ESC

Pierce123456, 2022-02-14:
I originally went to flash the bluejay firmware but it froze in the middle of flashing and eventually had to unplug it so i tried it again and couldnt get it. I finally got it to flash to ESC2,3,&4 but not ESC1 It wouldnt even read ESC1 and so I went to BLHELI Configurator to try and maybe reflash that firmware and after that none of the ESCs are working at all. It seems like its completely bricked. Is there a way I can unbrick it?

Pierce123456, 2022-02-16:
> Do the ESCs show up in the configurator at all? If they show up as
> "Unknown", then yes - you can flash them but you need to know the layout
> and check the "Ignore MCU Layout" before flashing.

Well 3 of the read and all three have bluejay on but something went wrong
with ESC 1 and it doesn't even read it anymore. All of the fets  on it look
good too, seems like all of the hardware is good. So there's no way I can
get ESC 1 to read again?

stylesuxx, 2022-02-16:
Sometimes the ESC might be read after leaving it plugged in for a bit longer: Connect the ESC, read it - when the 3 others show up, wait about a minute and try to read again, if you are lucky, it might show up again. Worst case you will need to re-flash the bootloader to this ESC, but this is a bit more involved and you will need to flash the MCU directly - this is not possible via the configurator.

stylesuxx, 2022-03-13:
@Pierce123456 any update on this issue?

stylesuxx, 2022-03-14:
OK, keep in mind that you can always attempt a flash via C2 interface: https://oscarliang.com/flash-blheli-c2-interface/

closed 2022-03-23 by stylesuxx