stylesuxx/esc-configurator #224: missing ESC and "Layout mismatch, override not enabled - aborted"

zakopanypouszy, 2022-02-16:
Hello, I flash my ESC and everything was fine, unfortunately during the test flight there was "desync" and since then one ESC completely stopped responding and disappeared from ESC-configurator, two more work normally but the third does not flash and I get a message " Layout revision 203 is not yet supported"

[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/8082723/esc-configurator-log.txt)

stylesuxx, 2022-02-16:
If you flash an unknown ESC you need to enable the Checkbox "Ignore MCU layout", then the flash will go through.

zakopanypouszy, 2022-02-16:
Thanks to BlHeli configurator and "Ignore MCU layout" "I was able to fix" unknow esc "but I still have a problem with one missing ESC ... connections look fine, voltage to ESC is coming, unfortunately, does not make any sound.

stylesuxx, 2022-02-16:
Yeah, if it does not show up in the configurator there is a couple things you could try:
* If your radio is on - switch it off, sometimes interference makes it hard to read
* Connect the ESC, read, wait for the ones that show up, to show up - then wait a bit - a minute should be OK, and read again. It might show up then

If none of the above works you might need to re-flash the bootloader to that specific MCU - if this does not work, the MCU of that ESC is probably gone.

stylesuxx, 2022-03-13:
@zakopanypouszy any update on this issue?

zakopanypouszy, 2022-03-22:
unfortunately I couldn't find a solution so I bought a new ESC

stylesuxx, 2022-03-23:
Hmm OK. There is still always the option to try and flash the MCU directly via C2 interface...

closed 2022-04-03 by stylesuxx