stylesuxx/esc-configurator #225: Después de usar la herramienta solo reconoce 3 esc!

Jota76, 2022-03-13:
He actualizado los esc con la versión blue jay 16 y solo 3 se han completado. el número 4 no lo reconoce, como lo puedo solucionar? he probado de varias formas y solo reconoce 3 esc

stylesuxx, 2022-03-13:
Sorry, please post in EN, otherwise I will not be able to help you.

Jota76, 2022-03-13:
I have updated the esc with the blue jay 16 version and only 3 have been completed. the number 4 does not recognize it, how can I fix it? Tried several shapes and only recognizes 3 esc

stylesuxx, 2022-03-13:
Does ist show up as "Unknown" or does the fourth one not show up at all?

Jota76, 2022-03-13:
It appears as

Unknown

stylesuxx, 2022-03-13:
Oh, then just re-flash the one that appears as unknown. You need to chose the proper layout from the dropdown and before flashing you need to click the checkbox "Ignore MCU and layout" and you should be good to go.

Jota76, 2022-03-13:
ok. Tomorrow I will try to see if it is solved and then comment, thanks

Jota76, 2022-03-14:
Hello, I have tried to return to flash and it has not worked. this is what appears to me. how is it solved? ,Thank you

Missing an ESC?
The amount of avallable ESCs dos not match what the flight controller is reportina. This can have multiple reasons:
The ESC was not ready yet. This can for example happen if it is playing a melody. Also a previous flash could have gone wrong and it needs longer to
There are simply not more ESCs connected.
The MCU of the ESC is defective.
If the result is not what you are expecting, try clicking the 'Read Setup' button again.
Save Debug Log
Clear Debug Log
Port utilization: D: 0% U: 0%
Packeterror
esc-configurator-.
Restore Default Setungs
Flash Al
Write Setun
ReadSetun

stylesuxx, 2022-03-14:
This happened after you tried to flash the Unknown ESC?

Try leaving it plugged in for a minute or so and read again, it might come back online.

Jota76, 2022-03-14:
No, it was already like this before flashing again
that indicates since I had the problem and it does not recognize it nor can I do anything with it. What I can do?
it is a fc aio and it is new it has little use and is in perfect condition

Jaime

El 14 mar 2022, a las 9:10, Chris ***@***.***> escribió:

This happened after you tried to flash the Unknown ESC?

Try leaving it plugged in for a minute or so and read again, it might come back online.

stylesuxx, 2022-03-14:
So yesterday you got "Unknown", today it does not show up at all? No other changes made?

As I say, let it connected for some time before reading it again, it might show up. Also turn off your radio should it be on.

Couple days ago I had someone who flashes a wrong Betaflight target and one of the ESCs disappeared because of that - might that be the case with you too?

Jota76, 2022-03-14:
the message is the same since I had the problem flashing the first time. nothing has changed, doing what you compared me yesterday.
It only recognizes 3 and the fourth does not let me modify it, it does not appear. what can i do?
I have had it connected for a while but it does not appear

excuse my english, i use translator

Jota76, 2022-03-14:
> Hmm, then your only option is to flash this ESC via C2 interface...

really? no choice? the fc is AIO, what is that interface where can I get it?

stylesuxx, 2022-03-14:
It's an Arduino that you flash with a specific sketch, I linked you to a tutorial on how to do it.

closed 2022-03-23 by stylesuxx