stylesuxx/esc-configurator #230: Unable to connect

dougjeff, 2022-03-31:
I was trying to connect and keep getting an error message that the port is in use, when it's not.
Using Edge browser Version 99.0.1150.55 (Official build) (64-bit)
Connecting to HGLRC mini f7 FC & 28A 4in1 ESC

stylesuxx, 2022-03-31:
Does the serial port selection come up?
Have you tried with a different Browser Like Chromium or Vivaldi?
Which windows are you on?
Is Betaflight Configurator working properly with your setup?
Have you double checked that no other configurator is running?

stylesuxx, 2022-04-07:
Hey @dougjeff, any update on this, could you provide me with the info I have asked for?

stylesuxx, 2022-04-07:
> I did respond to your email on March 31st.

Usually replies to github notifications show up here, it is github that sends the mails, not me. I unfortunately did not receive anything.

> I mentioned that the order that I connect seemed to make the difference.

What do you mean by that? Did you get it working in the end, or do you still need assistance?

dougjeff, 2022-04-08:
Hi Chris 

Sorry not to be clearer.  

The issue is resolved and I have Bluejay loaded :-) 

Thanks
Doug

 

Links:
------
[1]
https://github.com/stylesuxx/esc-configurator/issues/230#issuecomment-1091851966
[2]
https://github.com/

stylesuxx, 2022-04-08:
Hey @dougjeff that is great.
Could you tell me what resolved the issue in the end? Just in case someone runs into a similar problem.

dougjeff, 2022-04-09:
> Hey @dougjeff [1] that is great.

Hi Chris 

Sure 

The problem seems to be related to the order that I plug in the FC to
usb and when I plug in the battery. 

What worked in the end was to plug in the USB, connect to the
configurator and then to plug in the battery. 

Cheers 

Doug

 

Links:
------
[1] https://github.com/dougjeff
[2]
https://github.com/stylesuxx/esc-configurator/issues/230#issuecomment-1092521998
[3]
https://github.com/

stylesuxx, 2022-04-09:
That is very interesting...

Best practice is to do it the other way around, as to not to risk killing the USB port with a spike by attaching the battery later on. I am wondering if this can be reproduced on other computers - any way to check on a different machine?

dougjeff, 2022-04-10:
Hi Chris 

I will take it to work tomorrow to see if I get the same results there. 

Will let you know after work (I am on the west coast of Canada) 

Thanks
Doug

 

Links:
------
[1]
https://github.com/stylesuxx/esc-configurator/issues/230#issuecomment-1093927605
[2]
https://github.com/

stylesuxx, 2022-05-11:
closing due to inactivity.

dougjeff, 2022-10-11:
Hi Chris 

Thanks for your response. 

I feel a little stupid, but the problem appears to the the order that I
connect things. 

In the end I connected the battery to the quad and then plugged in and
hit connect on the ESC configurator.  

Originally I had plugged in the USB and tried to connect (thinking that
the FC was powered by USB and would enable the passthrough once I
connected the battery) 

Cheers 

Doug

 

Links:
------
[1]
https://github.com/stylesuxx/esc-configurator/issues/230#issuecomment-1084871716
[2]
https://github.com/