stylesuxx/esc-configurator #235: problem connect

dex1968, 2022-04-22:
Port already in use by another application - try re-connecting

dex1968, 2022-04-22:
2022-04-23 @ 00:32:25 -- Port already in use by another application - try re-connecting
2022-04-23 @ 00:32:24 -- Opened serial port
2022-04-23 @ 00:32:22 -- Port selected
2022-04-23 @ 00:31:40 -- Plugged in
2022-04-23 @ 00:25:13 -- Chrome (100.0.4896.127) on Windows

stylesuxx, 2022-04-22:
Could you please provide some more information:
* Which FC?
* Which Version of Windows?
* Are all other configurators closed?
* Can you connect via Betaflight Connfigurator?

dex1968, 2022-04-23:
good morning.
esc Hackrc 35a, changed bootloader "if I did the job right with cube32st"
arduino nano "configured with Blheli_32" two wires
windows 7
now that I changed the bootloader with the Blheli_32 configurator it no longer connects to the esc via arduino nano.
thanks for the help
Francis

stylesuxx, 2022-04-23:
Hey @dex1968 the configurator only supports flashing via Betaflight pass-through, so you will need an FC. alternative is to use the "native" AM32 configurator.

dex1968, 2022-04-23:
stlink?

stylesuxx, 2022-04-23:
No, ESC Configurator ONLY works with BF passthrough.

dex1968, 2022-04-25:
ok.with FC and passthrough it works.
pity not to be able to use other for single esc.

closed 2022-04-25 by stylesuxx