stylesuxx/esc-configurator #241: Mobula 7 ELRS throttle runaway

Mr-Vulcan, 2022-06-16:
I flashed my mobula 7 ecs which are ZH-30, to bluejay 0.16. Which worked. I set the min/max throttle to max which was said by another user who has the mobula 7. I then turned on bidectional in betaflight. However when I arm the motors 1 and 3 go off crazy and 2 and 4 stay low. However, when I add throttle 2 and 4 won't change and 1 and 3 go out of control no steady increase to my throttle level just increases to the max and then decreases its like a fluctuation. I have no clue how to fix this I've searched everywhere.

stylesuxx, 2022-06-16:
Same setup works with blheli_s?
Have you tried running without bi-directional DSHOT enabled?
Sounds to me as if something is wrong with the motor mapping if 2 and 4 don't do anything...

Anyway - I don't exactly see how this is a configurator issue? Maybe you could join the Bluejay discord, I am sure someone will be able to help you debug there.

Mr-Vulcan, 2022-06-16:
I am on each configurator discord but not bluejay. However I've tired with with bsheli which worked before I flashed the escs but after the bluejay flash. I reflashed back to bsheli and it's the same.

stylesuxx, 2022-06-16:
Yeah, then you must have changed something in Betaflight too. I suggest you flash to the initial state and try from there.

Mr-Vulcan, 2022-06-16:
I'll try, I've got it working slightly it can hold a idling throttle but trying to increase throttle causes it to cut off due to throttle runaway

stylesuxx, 2022-06-16:
Do the motors spin up in the motors tab?

Mr-Vulcan, 2022-06-16:
Yep the motors are perfect using the motors control tab able to control all of them without a issue. The only runaway I get is increasing throttle after arming

Mr-Vulcan, 2022-06-16:
I'll give that a try, I think all motors are spinning the same they are all giving off the same tone and the downdraft feels equal

stylesuxx, 2022-06-16:
You did double check motor rotation and props, right?

Mr-Vulcan, 2022-06-16:
Yep triple checked that! Currently reflashing the fc

Mr-Vulcan, 2022-06-16:
Just tired with the reflashed fc. Now the arming is not stable. Front right motor twitches, the back left spins slower than front left and back right.

Mr-Vulcan, 2022-06-16:
And only two out of 4 escs are being detected now

Mr-Vulcan, 2022-06-16:
It only shows 4 escs when bidirectional is turned off when its on only 2 is shown

stylesuxx, 2022-06-16:
bi-directional will not work with blheli_s. I would recommend you try everything completely stock and work from there.

Mr-Vulcan, 2022-06-16:
Im on bluejay when I enabled bi-directional which then showed only 2 escs, so ive turned it off and I just armed the quad and the idle was stable.

stylesuxx, 2022-06-16:
Yeah, there is definetly something fishy... You are running DSHOT300 I assume? What's your PID loop settings? Also might help to disable DSHOT bitbang in case it is enabled...

Mr-Vulcan, 2022-06-16:
I was running DSHOT300 however I just changed to DSHOT600 and it was more stable. Only issue now is front right motor half spinning It may be a faulty motor or esc for that one.

The image is the PIDS I havent touched them since the flashing
[image]

stylesuxx, 2022-06-16:
DSHOT300 is actually the recommendation for Bluejay. Why would you assume faulty motor or ESC, I thought it has been working before flashing around? Was this not the case? I would make sure to get everything running in the original state BF AND ESC firmware. Once you have the original - I assume, working state - move on from there.

Mr-Vulcan, 2022-06-16:
It did work before, my suspicion is just faulty ESC mainly I have a spare board I can try the motor on

stylesuxx, 2022-06-16:
if it's a faulty ESC you would see that easily in the motor tab. If it spins significantly lower than the others. Also flashing ESC's does not break them. So if you did not crash hard inbetween I'd say chances are slim. Although it's Happymodel hardware, so it would not surprise me if it just died randomly...

Mr-Vulcan, 2022-06-16:
I haven't had a hard crash. But I'll get in contact with happymodel so see if it may be a software bug in there fc software possibly

Mr-Vulcan, 2022-06-17:
All updates are on bluejay discord udner support

closed 2022-06-17 by Mr-Vulcan