stylesuxx/esc-configurator #244: BLHeli32 to RTTTL conversion

stylesuxx, 2022-07-12:
Use https://github.com/Globolo001/BLHELI_32_to_RTTTL to automatically convert BLHeli32 tunes to RTTTL, add some info about the editor being capable of that.

ltwin8, 2023-10-14:
> Use https://github.com/Globolo001/BLHELI_32_to_RTTTL to automatically convert BLHeli32 tunes to RTTTL, add some info about the editor being capable of that.

Could you please give an example maybe based on the song in videos description? Like the variables octave, length, interval… I can not find octave…

https://youtu.be/CqF1gtB6Txg?feature=shared

best regards

stylesuxx, 2023-10-14:
You should probably ask the creator of the converter. I have no idea about Blheli32 data structures at all.

stylesuxx, 2024-07-26:
Closing since BLHeli32 is no longer a thing.

closed 2024-07-26 by stylesuxx