stylesuxx/esc-configurator #247: AM32 flashing fails with local files

howels, 2022-08-05:
**Describe the bug**
Cannot flash AM32 local files

**To Reproduce**
1. Read ESC configuration.
2. ESCs are shown
3. Select ESC1 and select flash
4. Select local file, check "ignore inappropriate MCU type" in case this might stop flashing
5. Click Flash
6. Nothing happens - returned to main page

Occurs whether attempting to flash all ESCs or individually. Had to use the AM32 local application to flash these local files.

File type and format checks out because I can flash these with the local AM32 tool.

**Expected behavior**
Should flash the file or produce an error clearly stating why it's not possible (wrong MCU type etc)

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser: Chrome latest 103
 - Version 0.23? (current 2022-08-04)

**Log**

```
2022-08-04 @ 11:02:14 -- Failed flashing ESC 3 - check file type
2022-08-04 @ 11:02:14 -- Flashing ESC 3
2022-08-04 @ 11:02:14 -- Failed flashing ESC 2 - check file type
2022-08-04 @ 11:02:14 -- Flashing ESC 2
2022-08-04 @ 11:02:14 -- Failed flashing ESC 1 - check file type
2022-08-04 @ 11:02:14 -- Flashing ESC 1
2022-08-04 @ 11:01:56 -- Failed flashing ESC 1 - check file type
2022-08-04 @ 11:01:56 -- Flashing ESC 1
2022-08-04 @ 11:01:41 -- Failed flashing ESC 4 - check file type
2022-08-04 @ 11:01:41 -- Flashing ESC 4
2022-08-04 @ 11:01:41 -- Failed flashing ESC 3 - check file type
2022-08-04 @ 11:01:41 -- Flashing ESC 3
2022-08-04 @ 11:01:41 -- Failed flashing ESC 2 - check file type
2022-08-04 @ 11:01:41 -- Flashing ESC 2
2022-08-04 @ 11:01:41 -- Failed flashing ESC 1 - check file type
```

howels, 2022-08-05:
Maybe related - the esc-configurator tool shows an incorrect EEPROM revision, which doesn't match the firmware reported in AM32's config tool or the file:
[image]

```
2022-08-04 @ 11:01:41 -- Flashing ESC 1
2022-08-04 @ 11:01:24 -- Done reading ESCs
2022-08-04 @ 11:01:24 -- Read ESC 4: GD32Pb4 - AM32, 243.16, Bootloader v1 (PB4)
2022-08-04 @ 11:01:24 -- Firmware mismatch! Flash: 1.85 vs. EEPROM: 243.16
2022-08-04 @ 11:01:24 -- Read ESC 3: GD32Pb4 - AM32, 243.16, Bootloader v1 (PB4)
2022-08-04 @ 11:01:24 -- Firmware mismatch! Flash: 1.85 vs. EEPROM: 243.16
2022-08-04 @ 11:01:24 -- Read ESC 2: GD32Pb4 - AM32, 243.16, Bootloader v1 (PB4)
2022-08-04 @ 11:01:24 -- Firmware mismatch! Flash: 1.85 vs. EEPROM: 243.16
```

howels, 2022-08-05:
Apparently the version number is in EEPROM, but esc-configurator is looking for a fixed location in the ROM and that is causing the issue because different MCU targets in AM32 use different memory locations for their data.
The EEPROM version number should be used according to @AlkaMotors

howels, 2022-08-05:
Maybe the version number thing is different to the original issue - cannot flash local firmware. But it may be linked because this is a new target MCU I am trying to flash (GD E230).

stylesuxx, 2022-08-05:
OK, careful then. If the EEPROM layout changed then writing settings might create problems anyway. I am a bit surprised that you don't get a warning about they layout not being recognized...

PR #250 (2022-08-17): Feature/247
Should fix #247
closed 2022-08-31 by stylesuxx