stylesuxx/esc-configurator #248: Can't update FW stuck on blhelis 16.5

tomarcom, 2022-08-05:
[esc-configurator-log (1).txt](https://github.com/stylesuxx/esc-configurator/files/9273690/esc-configurator-log.1.txt)

stylesuxx, 2022-08-09:
I see multiple successfull flashes in the log and one where something did not seem to work out.
Which FC is the one in question that does not work? I can see it has a rather old version of BF flashed - 4.1.5

Can you update this esc with the standolone BLHeli_s configurator?
Are you sure that the battery did not go flat during  flash? The log ends so suddenly.

tomarcom, 2022-08-09:
I'm sorry I did not know there were several in the same file.  The MatekF11 "TinyHawk" flashed to blhelis 16.5, but it locked the ESC to that version.  No other configurator works, the ESC just reboots upon attempting to reflash it.  Seems like the only fix is to flash using Arduino/blheli configurator from the C2 pads, which I can't seem to properly identify.  I'm not sure why the esc web configurator "Locked" the ESC's but it did it on 2 different MatekF411's and a NBD brushless hummingbird pro.  A 3 now "Locked" on blhelis 16.5.  I'm not looking for a resolution, just wanted ya'll to be aware of the issue.  Thanks, Tracy

tomarcom, 2022-08-09:
> So you used the web configurator to flash to BLheli_S 16.5 - from which initial version. And you are telling me that after flashing to 16.5 it is locked? Just out of curiosity, why were you flashing to 16.5 in the first place? Can you still change settings?
>
> The log is telling a different story, that's why I am so surprised, I don't see an attempt where you flashed to BLHeli_S 16.5. Only where you attempt to flash Bluejay v0.16 to an ESC already flashed with BLHeli_s 16.5...
>
> I have a couple of TinyHawks myself and can't reproduce this at all, so I would really be curious on your exact steps you took. Especially at which firmware you started out on.

I first updated to BlueJay.16 then the qwad wouldn't spin the all the motors, so I rolled back to an old version without bidirectional d-shot, and 24k "16.5"  The issue turned out to be the tune and not the esc's anyway, but they were now locked on 16.5.  I'd be more than happy to try anything you suggest or send you more logs.Tracy

stylesuxx, 2022-08-09:
OK so you went from:

```
(0) unknown version -> (1) Bluejay 0.16 -> (2) BLHeli_S 16.5 [Locked]
```

Can you send me the log where you went from (0) to (1) please? I guess this must have happened on a different computer? Otherwise it would be part of the logs that you already sent me.

tomarcom, 2022-08-09:
Yes you are so right, I do have a laptop that was also used.  Attached is the log from it.

 
 

  
2022/08/09 14:41:32: Chrome (103.0.0.0) on Mac OS
2022/08/09 14:43:32: Port selected
2022/08/09 14:43:35: Opened serial port
2022/08/09 14:43:35: MultiWii API version received: 1.44.0
2022/08/09 14:43:35: Flight controller info, identifier: BTFL version: 4.3.1
2022/08/09 14:43:35: Running firmware released on: Jul 13 2022 03:32:52
2022/08/09 14:43:35: Board: S411, version: 0
2022/08/09 14:43:35: Unique device ID received - 0x053003c3437510c37313935

stylesuxx, 2022-08-09:
Hmm, I am afraid that github does not attach files if you answer via E-Mail, I am afraid you will add it via browser.

tomarcom, 2022-08-09:
[esc-configurator-log (2).txt](https://github.com/stylesuxx/esc-configurator/files/9294274/esc-configurator-log.2.txt)

tomarcom, 2022-08-09:
Sorry it looks like I cleared the log from the laptop.  I uploaded it anyway.

closed 2022-09-06 by stylesuxx