stylesuxx/esc-configurator #249: AM32 G071

stylesuxx, 2022-08-08:
To make the g071 esc's compatible there are only two changes. The identifier : https://github.com/AlkaMotors/g071Bootloader/blob/f86fc79a05c7ddcc25d598f4b2c952204c98c674/Core/Src/main.c#L62

and eeprom location https://github.com/AlkaMotors/g071Bootloader/blob/f86fc79a05c7ddcc25d598f4b2c952204c98c674/Core/Src/main.c#L18

AlkaMotors, 2022-10-15:
@stylesuxx I have a couple mac users who would love to get this working. Those are the only changes that to make the g071 work. The eeprom layout is the same. Since there are now more and more arm mcu options, all esc's with 32 memory will report as the f051 mcu and 64 KB esc's will use the x3506 identifier. The page size can vary from 1kb to 2kb depednign on the mcu but the writing can be done in 1kb page size for all.  

Maybe something like this?

 `{
      "name": "ARM64K",
      "signature": "0x3506",
      "page_size": 1024,
      "flash_size": 65536,
      "flash_offset": "0x08000000",
      "firmware_start": "0x1000"
    }`

As far as i can see the only line that would need to be changed depending on the identifier would be this ?
the offset for all 64k will be 0xF800.
 
https://github.com/stylesuxx/esc-configurator/blob/21aa35d55b7b86231adbaf42c747b46d9034d04f/src/sources/AM32/eeprom.js#L9

stylesuxx, 2022-10-15:
Hey @AlkaMotors - that is perfect timing! I had to change eeprom offset for the BB51 too and was wondering if I should make the same changes to the AM32 config. I will do this either later today or tomorrow and you can give it a spin.

stylesuxx, 2022-10-17:
@AlkaMotors I published the adaptations on the preview build, could you let me know if it works with the 64k? https://preview.esc-configurator.com/

ljlzonda, 2022-10-18:
hi bro, just did some test with my G071 ESC setup, and it works fine for me!

stylesuxx, 2022-10-18:
@ljlzonda awesome, thanks for the feedback!

ljlzonda, 2022-10-18:
> @ljlzonda awesome, thanks for the feedback!

settings all work out fine, but I found two missing settings comparing to standalone AM32 configurator , Sine Mode Power and Running Brake Level, also reported this issue with @AlkaMotors

closed 2022-10-24 by stylesuxx