stylesuxx/esc-configurator #261: Cannot connect (Port already in use by another application - try re-connecting)

StillVoidingWarranties, 2022-09-16:
ESC Configurator refuses to connect.
FC: Mamba F405 Mini MK3
Browser: Chromium
OS: Pop!_OS 20.04
Log messages:
2022-09-17 @ 01:18:09 -- Port already in use by another application - try re-connecting
2022-09-17 @ 01:17:51 -- Plugged in
2022-09-17 @ 01:17:51 -- Chrome (83.0.4103.116) on Linux

FC connects fine to Betaflight Configurator and Bluejay Configurator Standalone.

stylesuxx, 2022-09-16:
How did you install Chromium? I remember seeing such problems in some circumstances when Chromium was installed via snap...

Would be interesting to see if it works on a "native" Chrome install.

StillVoidingWarranties, 2022-09-17:
Chromium was installed from the Ubuntu repository (Focal) (Pop!_OS is largely compatible with Ubuntu and uses deb packages).
Is there anything I could do to help getting debug info on this?

stylesuxx, 2022-09-17:
Yes, please first see if it works with Chrome on your machine, just that we know if it generally works on any browser on your machine. If it does we can look into Chromium.

Also check ```chrome://device-log/``` on Chromium and see if something suspicious is visible.

StillVoidingWarranties, 2022-09-17:
Thank you. Original Chrome worked ok, though I wouldn't love to have to use Chrome.

`2022/09/17 19:41:09: Chrome (105.0.0.0) on Linux
2022/09/17 19:41:09: Plugged in
2022/09/17 19:41:21: Chrome (105.0.0.0) on Linux
2022/09/17 19:41:28: Port selected
2022/09/17 19:41:32: Opened serial port
2022/09/17 19:41:32: MultiWii API version received: 1.44.0
2022/09/17 19:41:32: Flight controller info, identifier: BTFL version: 4.3.1
2022/09/17 19:41:32: Running firmware released on: Jul 13 2022 03:32:11
2022/09/17 19:41:32: Board: S405, version: 0
2022/09/17 19:41:32: Unique device ID received - 0x0310042374e501220353355
2022/09/17 19:41:42: Trying to read 4 ESCs
...
2022/09/17 19:41:43: Read ESC 4: G-H-15 - Bluejay (BETA), 0.14, 48kHz
2022/09/17 19:41:43: Done reading ESCs`

If Chromium is not supposed to be supported then this issue is resolved. I will play around to find clues why Chromium behaves differently.

stylesuxx, 2022-09-17:
Chromium is supported, lot's of people use it. It seems to have issues in some cases.
Please check the link to the other (resolvess) Chromium issue posted before.

Check the chromium device log and post exact version. I can try and see if I can re-create the issue with your version.

StillVoidingWarranties, 2022-09-17:
Just a quick datapoint: Chrome 105, Chromium 105 and Chromium 85 show different devices in the open port selection dialog.

Chrome 105: ttyACM0 (works), ttyS0, ttyS1, ...
Chromium 105: ttyS0, ttyS1, ... (doesn't work)
Chromium 85: Betaflight STM32F405 (doesn't work)

I will read/investigate.

StillVoidingWarranties, 2022-09-17:
It works now. After closing Chromium and unplugging USB, then plugging the FC in, waiting 10 seconds and only then then starting Chromium and opening esc-configurator the ttysACM0 device was detected and I could connect without problems. The delay before opening esc-configurator seems to be crucial. Thank you for your support.

stylesuxx, 2022-09-17:
Very interesting, thanks for testing and coming back with this info - this is repeatable and reproducible? And this happened with both Chromium versions you mentioned above?

StillVoidingWarranties, 2022-09-17:
I only tested with the 105 version of Chromium because I uninstalled the 85 version before. I will try to reproduce the effect and report back. BTW the exact version of Chromium is 105.0.5195.125 (Official Build) (64-Bit) as flatpack.

stylesuxx, 2022-10-07:
@StillVoidingWarranties any update on this, could your reproduce? If not I will add this to the known issue section and close the issue...

stylesuxx, 2022-10-18:
Closed due to inactivity. Feel free to update the ticket with additional information and re-open.

closed 2022-10-18 by stylesuxx