stylesuxx/esc-configurator #266: **FLASH*FAILED** version '0.16' (Layout 204) is not yet supported.  Let us know by opening an issue

FlyinRyanFPV, 2022-10-15:
Good evening,
 I am Trying to flash a GEPRC GEP-F11-35A AIO with BLUEJAY at 48K PWM but ESC #1 failed the flash with this error "**FLASH*FAILED** version '0.16' (Layout 204) is not yet supported". Attached is the Log File for reference. 

Please advise?

Thanks,
Ryan Petroski
[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/9790990/esc-configurator-log.txt)

stylesuxx, 2022-10-15:
Flash again, just this one ESC. You will have to select the correct layout and check the "ignore MCU layout" checkbox. Make sure your radio is off and your battery is full.

FlyinRyanFPV, 2022-10-17:
I finally got it! Had to reflash Betaflight for some reason. Thank you for all the help!!
Cheers 🍻

Ryan Petroski
Direct 832-616-0290
3646 Fir Forest Dr
Spring, TX 77388

stylesuxx, 2022-10-17:
Hmm, interesting. Which version of Betaflight were you on and to which version did you upgrade?

FlyinRyanFPV, 2022-10-18:
I'm not sure what the deal was! It was on 4.3 so just re-flashed 4.3. when I flashed the #1 ESC on the AIO (as per your instructions)  it went fine but when I tried to do the #2 ESC, that originally had no issues before, failed and on the second attempt it completely disappeared after two failed attempts.  After a dozen of booting it up, plugging in, unplugging it and trying to reflash it I finally was able to get #2 back but then the other two ESC's started showing up as "unavailable" so that's when I re-flashed BF, afterwards I was able to flash all ESC's at one time with no issues. I believe it was a hardware problem but IDK. I didn't see any physical damage as it was a brand new AIO.🤷🏻
Thanks,Ryan

stylesuxx, 2022-10-18:
OK, thank you. Sounds very random and as if the ESC simply showed up after being powered some time. This might happen, depending in which state the flash failed. But technically there is nothing I can do after the fact.

closed 2022-10-18 by stylesuxx

FlyinRyanFPV, 2022-10-18:
Just want to say again, thank you for all your help! She's running tip-top now with no issues. 
Cheers friend!

Ryan Petroski
Direct 832-616-0290
3646 Fir Forest Dr
Spring, TX 77388

Aquaqx, 2023-12-17:
> Flash again, just this one ESC. You will have to select the correct layout and check the "ignore MCU layout" checkbox. Make sure your radio is off and your battery is full.

Had the same issue as Ryan. This totally worked. Thanks a lot!