stylesuxx/esc-configurator #268: Trying to flash Meteor75 with BlueJay "check file type" caused by 404 for hex link

jppresents, 2022-10-17:
I am trying to flash Meteor75 (WalkSnail) to BlueJay - I use preview.esc-configuratior.com (otherwhise settings can't be read) - but when I try to flash, I get "Failed flashing ESC 4 - check file type" on all ESC.

From the Debug Log: 

2022/10/17 02:03:42: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.16/A_X_5_96_v0.16.hex 
2022/10/17 02:03:42: Flashing ESC 1
2022/10/17 02:03:42: Failed flashing SiLabs: TypeError: Cannot read properties of null (reading 'data')
[... same for alle esc ...]

Turns out, the chosen firmware link gives a 404 not found.

(This is using https://preview.esc-configurator.com/ - because normal url can not read the information from the quad.)

stylesuxx, 2022-10-17:
You need to download the latest [TESTING assets](https://github.com/bird-sanctuary/bluejay/pull/12#issuecomment-1273842356) and flash from "local" if you want to try the preview version for X MCUs.

jppresents, 2022-10-17:
That did work.
I now see Version 17 BETA when reading the information.
But I have to report, that reading from the ESCs is now unreliable and so is writing (motor reverse).
Both only works some time. I end up with not 4 but randomly 1 through 4 ESCs read. (An the message that some are missing).
And while writing it also fails often.

2022-10-17 @ 03:25:25 -- Done reading ESCs
2022-10-17 @ 03:25:25 -- Failed reading ESC 4
2022-10-17 @ 03:25:11 -- Failed reading ESC 3

2022-10-17 @ 03:24:12 -- No changes - not updating ESC 3
2022-10-17 @ 03:24:12 -- Failed writing settings to ESC 1
2022-10-17 @ 03:23:53 -- Done reading ESCs

Like this.

Managed to set my motor direction correctly now - so I will let this go for now and mark this as closed.

stylesuxx, 2022-10-17:
Please attach the complete log (bottom left, download log) after enabling extended logging. and running the write/save operation.

The new firmware does not change how read/write works and it is most likely an issue on your end (eg bad connection or radio on while reading/writing), I would still like to see the logs please in case I am overlooking something here.