stylesuxx/esc-configurator #272: No bluejay firmware found for O-H-5 layout

PSi86, 2022-10-20:
Hi Team,
The esc-configurator currently shows no available bluejay firmware for a whoop board with O-H-5 layout. O-H-5 is correctly detected, but you cannot select a firmware version. When selecting BLHeli-S it shows available firmware versions correctly. I have bluejay on the esc already so in the past it must have been possible to flash bluejay to this esc. Or is there a special reason to remove the support for this esc layout?

Thank you!

stylesuxx, 2022-10-20:
Proabably something went wrong when fetching the version list for Bluejay.
> Right click on the page -> inspect -> Application -> Cache storage -> v1
Whhich entries do you see there? When in doubt you can also just delete them and reload the page.

PSi86, 2022-10-20:
There was the old BlueJay Repo still in v1 Cache storage. After deletion it works now. Thank you a lot!

closed 2022-10-20 by PSi86