stylesuxx/esc-configurator #327: Cannot connect to matek H743 wing, saying Port already in use by another application - try re-connecting

lotsofone, 2023-02-14:
**Describe the bug**
When I press connect, it says "Port already in use by another application - try re-connecting"

**Screenshots**
[image]

**Log**
```
2023-02-14 @ 10:59:27 -- Port already in use by another application - try re-connecting
2023-02-14 @ 10:59:26 -- Opened serial port
2023-02-14 @ 10:59:25 -- Port selected
2023-02-14 @ 10:59:22 -- Port already in use by another application - try re-connecting
2023-02-14 @ 10:59:21 -- Opened serial port
2023-02-14 @ 10:59:18 -- Port selected
2023-02-14 @ 10:59:08 -- Plugged in
2023-02-14 @ 10:59:08 -- Chrome (109.0.0.0) on Windows
```

**Desktop (please complete the following information):**
 - OS: Win11
 - Browser chrome
 - Version 109.0.0.0

**Additional context**
It is okay to connect and flash ESC by Matek F405-WMN, but not okay by Matek H743 wing.
I noticed that Matek H743 wing has 2 com ports when connected to PC.
[image]

While Matek F405-WMN only has one.
[image]

Could this be the reason?

stylesuxx, 2023-02-14:
So, when selecting the port, do multiple com ports show up? The message usually means that some other application is claiming the serial port.

Make sure that no other configurator is running and close all other apps that might be claiming the serial port. Cura for example is known to do that...

lotsofone, 2023-02-14:
When I plug in the controller without battery, it first shows up COM5
[image]
Then comes COM6 and COM7, and COM5 disappers
[image]
Both the two ports give the same error "Port already in use".
I am pretty sure there is no others claiming the port, because I can connect to FC with MP later.

stylesuxx, 2023-02-14:
Not entirely sure what could be going on here - I would assume that the first one is the bootloader device that disappears ofter some time. Not sure why there then would be two serial devices. You are running ardupilot on this FC, right? You can connect with the original BLHeli_S configurator?

Please also post output of `chrome://device-log/` - clear it first, then attach the flight controller and post a screenshot of the output.

lotsofone, 2023-02-14:
Yes I am running ardupilot on my FCs. I didn't try original BLHeli_S configurator, only tried the web one. I can connect to the ESC through Matek F405-WMN FC, but cannot through Matek H743 Wing FC.
When I plug in the Matek H743 Wing without battery, the log is
[image]

For Matek F405-WMN, the log is
[image]

stylesuxx, 2023-02-14:
Hmm, yeah - can't see an issue there. Try the original BLHeli_S configurator please and see if you can connect with that.

You have the same ardupilot versions on both FCs I assume?

lotsofone, 2023-02-14:
Both FCs have Plane 4.3.3
Is this the original configurator? https://github.com/blheli-configurator/blheli-configurator/releases/tag/1.2.0
It can connect to Matek F405-WMN, can't connect to Matek H743 Wing.
The error message is a little different.
[image]

stylesuxx, 2023-02-14:
Well, then it's not an issue of the configurator, you will have to ask Matek or Ardupilot how you are intended to connect to the ESCs.

closed 2023-02-14 by stylesuxx

stylesuxx, 2023-02-14:
Cool, then please report your findings to ardupilot, maybe they can fix it.