stylesuxx/esc-configurator #330: Read set up issue with 2 esc

MFS-Xavier, 2023-02-21:
Hello 

I built a Y6 ( 6 motors, 2 esc tmotor proii 55a and 1x FC TMOTOR F7 PRO).

I can read the 6 esc fine with blheli32 but I want bluejay on all my Quads.

I can connect to the drone with bluejay web interface.
I can then control the 6 motors....
But when I try to read setup.

I get the following error.  
the amount of available escs does not match what the flight controller is reporting....

My drone is powered. I just tried woth blheli32 before and it worked... so I m guessing a bluejay issue.

Can you help me? 

Thank you.

Regards

stylesuxx, 2023-02-21:
Bluejay does not work on BLheli_32 ESCs. Generally BLHeli_32 is not supported by the configurator.

closed 2023-02-21 by stylesuxx

MFS-Xavier, 2023-02-21:
Thank you

Just noticed it.
Sorry for wasting your time

Xavier POELMANS

stylesuxx, 2023-02-21:
Nothing to be sorry for ;-)

In case you want to liberate your ESCs you should look into AM32 - an open source firmware for 32 bit ESCs.

MFS-Xavier, 2023-02-21:
Well...i ahould have  read more before bithering you. Thank you so much though..
As for arm32, iwas just reading about this now.
I ll most likely go down that road.
Thank you
Xavier POELMANS