stylesuxx/esc-configurator #344: Cant Flash with Online Tool

DrJohannessen, 2023-03-19:
So with this stack https://www.speedybee.com/speedybee-f7-mini-35a-3-6s-8-bit-flight-controller-stack/ i wasnt able to flash the escs online. it just didnt show them on the right side. i then flashed them with the standalone configurator wich worked, after that flashing online worked again. The escs are jh40s, lemme know what information you need.

stylesuxx, 2023-03-19:
Please post the log. Would would exactly not work? If it works now, it probably would also have worked before. Something must have been different when you initially tried.

2xWhiskey, 2023-03-23:
I am currently unable to flash as well where I have been able to before when Bluejay was at 0.18.  (Sub250 AIO, Redfox A1 F4 5A) 
The Log:

2023-03-23 @ 12:55:29 -- Could not get file for flashing.
2023-03-23 @ 12:53:51 -- Done reading ESCs
2023-03-23 @ 12:53:51 -- Read ESC 4: O-L-5 - Bluejay, 0.18, 48kHz
2023-03-23 @ 12:53:50 -- Read ESC 3: O-L-5 - Bluejay, 0.18, 48kHz
2023-03-23 @ 12:53:50 -- Read ESC 2: O-L-5 - Bluejay, 0.18, 48kHz
2023-03-23 @ 12:53:50 -- Read ESC 1: O-L-5 - Bluejay, 0.18, 48kHz
2023-03-23 @ 12:53:49 -- Trying to read 4 ESCs
2023-03-23 @ 12:53:45 -- Unique device ID received - 0x04400734247500320393534
2023-03-23 @ 12:53:45 -- Board: S411, version: 0
2023-03-23 @ 12:53:45 -- Running firmware released on: Mar 14 2023 01:14:40
2023-03-23 @ 12:53:45 -- Flight controller info, identifier: BTFL version: 4.4.0
2023-03-23 @ 12:53:45 -- MultiWii API version received: 1.45.0
2023-03-23 @ 12:53:45 -- Opened serial port
2023-03-23 @ 12:53:39 -- Plugged in
2023-03-23 @ 12:53:21 -- Chrome (111.0.0.0) on Windows

stylesuxx, 2023-03-23:
@2xWhiskey support for the L MCU has been dropped. You will have to stay on v0.18.

2xWhiskey, 2023-03-23:
> @2xWhiskey support for the L MCU has been dropped. You will have to stay on v0.18.

Thank you for the response.  I also find that I cannot reflash v0.18 at all (so I can test different kHz versions).  Is that support intentionally dropped as well?

stylesuxx, 2023-03-23:
No, that should still be working. What kind of error do you get when trying to flash to v0.18?

2xWhiskey, 2023-03-23:
I get the same exact log messages as I pasted previously.  Being a layman, It seems to me like the file download link isn't working for the configurator.  Maybe the address was changed recently?

stylesuxx, 2023-03-23:
The files are all there. Please enable extended logging and post the full log (download on bottom left).

2xWhiskey, 2023-03-23:
> **EDIT**: Although - 96kHz is not available for L MCUs, were you trying to run that?

Flashing to 96kHz was exactly what I was trying to do, I'll settle for 48kHz v0.18.  Thank you for your time sir!