stylesuxx/esc-configurator #373: Is there anyway we can get "Sine Mode Power" And "Running Brake Level" migrated over to this point of access, rather than just the "Multi_ESC_Configurator" computer based program? Thanks!

Muddauber7, 2023-05-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

stylesuxx, 2023-05-06:
Sorry, I have no idea what you are talking about - can you provide more context please? Is this a setting you are looking for in AM32?

Muddauber7, 2023-05-06:
> Sorry, I have no idea what you are talking about - can you provide more context please? Is this a setting you are looking for in AM32?

Yes, these are available in our 1.82 version of the computer based programmer we're using for AM32.
Bottom right corner shows the additional settings that benefit the crawler side of all this.

[image]

stylesuxx, 2023-05-07:
@AlkaMotors could you please provide more infos about those settings? Short description of what they do, their value range and their position in the EEPROM? I'll do the rest then.

Muddauber7, 2023-05-07:
> @AlkaMotors could you please provide more infos about those settings? Short description of what they do, their value range and their position in the EEPROM? I'll do the rest then.

I've reached out to him on Discord. 
Additionally, is there a reason this doesn't work on our phones, and only works on desktop? 
Even when using the phone in desktop mode?

stylesuxx, 2023-05-07:
> Additionally, is there a reason this doesn't work on our phones, and only works on desktop? Even when using the phone in desktop mode?

It works on some phones - works fine on mine for example, generally speaking this is an issue in Android that is being addressed for some years: https://github.com/stylesuxx/esc-configurator/issues/56 Nothing I can do about it really.

Muddauber7, 2023-05-07:
Ok, thank you for the super fast response.
Hopefully we can get the additional settings added once you have what you need. 
Appreciate it.

AlkaMotors, 2023-05-08:
Running brake level is byte offset 42, range 0 to 10.

This is the amount of braking used when the motor is running. 

(Currently only working on the f051 and g071 and f031 mcu's)

Sine mode power is byte offset 45, range 1 to 10 
This is the fixed duty cycle level to be used when using sine mode. 

This have a warning not to go over 6 without testing and checking temperatures. For most motors a setting above 8 will either destroy the ESC or the motor!

stylesuxx, 2023-05-08:
Thanks @AlkaMotors - this is all available from v1.95 going forward I assume?

What happens if those settings are set on other MCUs - will they simply be ignored? Currently there is no distinction in EEPROM layouts for different MCUs. If they are going to diverge, I need to think about how to best solve it. If the settings are being ignored, I just add this info to the Help Text.

stylesuxx, 2023-05-09:
Merged into develop, available here: https://develop.esc-configurator.com/

closed 2023-05-09 by stylesuxx

Muddauber7, 2023-05-14:
Apologies for the Noob here, but this being closed, does that mean it's complete with the feature request?
Seems the settings are not there yet. And flash firmware is still broken, fails every time.

stylesuxx, 2023-05-15:
@Muddauber7 did you try it on the develop link I posted above? 

Flashing firmware failing you are just now mentioning for the first time (I think?). Please also check on the develop link and post a log of your flashing attempt in a separate issue please.