stylesuxx/esc-configurator #382: Rpm filter error on Speedybee ESC mini

ciscoarizona, 2023-05-26:
This forced me to turn off bidirectional d shot

stylesuxx, 2023-05-26:
Could you be more specific about what is happening? Also how is this an issue with the configurator? Sounds more like a Bluejay problem, if at all. Anyway, please post more information, FC firmware and version, bluejay Version and especially the specific error you are experiencing.

ciscoarizona, 2023-05-26:
Chris,

I am using the latest version of Bluejay from the esc-configurator.com  I
am also running BF 4.4.  When I try to arm, I get "RPM FILTER" error on my
OSD warning.  I suspect it's a bluejay problem, I just wanted to make
sure.
Cisco

stylesuxx, 2023-05-27:
I still need more information here - if you spin the motors up in Betaflight motors tab, do you see errors there too? I assume you are running DSHOT 300, right? What are your pid-loop settings - can you please share a dump of your Betaflight settings? Also please link the exact flight controller/esc combo you are using.