stylesuxx/esc-configurator #383: Cant Read ESC's: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm

wolfgangteves, 2023-05-31:
Dear Developers of ESC Configurator,

i experience a problem where i can start the motors using esc configurator but not read the configuration. The Debug Log sais UnknownPlatformError.

On the same computer (M1 Mac) i can successfully read the firmware of a drifferent quad.

Would greatly appreciate any input.

best regards

```
2023/05/31 12:19:58: Chrome (113.0.0.0) on Mac OS
2023/05/31 12:19:58: Plugged in
2023/05/31 12:20:01: Opened serial port
2023/05/31 12:20:01: Opened serial port
2023/05/31 12:20:01: MultiWii API version received: 1.45.0
2023/05/31 12:20:01: Flight controller info, identifier: BTFL version: 4.4.0
2023/05/31 12:20:01: Running firmware released on: Apr 11 2023 06:28:22
2023/05/31 12:20:01: Board: S7X2, version: 0
2023/05/31 12:20:01: Unique device ID received - 0x02b00215632500120303236
2023/05/31 12:20:11: Trying to read 4 ESCs
2023/05/31 12:20:11: ESC 1 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:11: Failed reading ESC 1
2023/05/31 12:20:11: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:11: ESC 2 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:11: Failed reading ESC 2
2023/05/31 12:20:11: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:11: ESC 3 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:11: Failed reading ESC 3
2023/05/31 12:20:11: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:11: ESC 4 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:11: Failed reading ESC 4
2023/05/31 12:20:11: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:13: Trying to read 4 ESCs
2023/05/31 12:20:13: ESC 1 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:13: Failed reading ESC 1
2023/05/31 12:20:13: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:13: ESC 2 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:13: Failed reading ESC 2
2023/05/31 12:20:13: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:13: ESC 3 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:13: Failed reading ESC 3
2023/05/31 12:20:13: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:13: ESC 4 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:13: Failed reading ESC 4
2023/05/31 12:20:13: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:14: Trying to read 4 ESCs
2023/05/31 12:20:14: ESC 1 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:14: Failed reading ESC 1
2023/05/31 12:20:14: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:14: ESC 2 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:14: Failed reading ESC 2
2023/05/31 12:20:14: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:14: ESC 3 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:14: Failed reading ESC 3
2023/05/31 12:20:14: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:14: ESC 4 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:14: Failed reading ESC 4
2023/05/31 12:20:14: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:36: Trying to read 4 ESCs
2023/05/31 12:20:36: ESC 1 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:36: Failed reading ESC 1
2023/05/31 12:20:36: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:36: ESC 2 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:36: Failed reading ESC 2
2023/05/31 12:20:36: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:36: ESC 3 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:20:36: Failed reading ESC 3
2023/05/31 12:20:36: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:20:36: ESC 4 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
[... 48 lines cut ...]
2023/05/31 12:21:56: Closed port
2023/05/31 12:21:57: MSP command failed: Timed out after 1000ms
2023/05/31 12:22:03: Unplugged
2023/05/31 12:24:28: Chrome (113.0.0.0) on Mac OS
2023/05/31 12:24:46: Plugged in
2023/05/31 12:24:49: Opened serial port
2023/05/31 12:24:49: MultiWii API version received: 1.45.0
2023/05/31 12:24:49: Flight controller info, identifier: BTFL version: 4.4.0
2023/05/31 12:24:49: Running firmware released on: Apr 11 2023 06:28:22
2023/05/31 12:24:49: Board: S7X2, version: 0
2023/05/31 12:24:49: Unique device ID received - 0x02b00215632500120303236
2023/05/31 12:25:07: Trying to read 4 ESCs
2023/05/31 12:25:07: ESC 1 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:25:07: Failed reading ESC 1
2023/05/31 12:25:07: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:25:07: ESC 2 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:25:07: Failed reading ESC 2
2023/05/31 12:25:07: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:25:07: ESC 3 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:25:07: Failed reading ESC 3
2023/05/31 12:25:07: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/05/31 12:25:07: ESC 4 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/05/31 12:25:07: Failed reading ESC 4
2023/05/31 12:25:07: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
```

stylesuxx, 2023-05-31:
Well, this sounds like you are using an ESC that is not supported by ESC-configurator. Can you please tell me which ESC you are using exactly? Please link to a product page.

wolfgangteves, 2023-05-31:
Im sorry, youre right! i'm using blheli 32. so sorry for opening this.

https://www.copterfarm.de/iflight-succex-d-mini-f7-twing-stack-mini-f7-45a-esc-1021.html