stylesuxx/esc-configurator #403: ESC-Configurator not working

Teklock, 2023-07-28:
2023/07/28 05:27:43: Edge (115.0.1901.183) on Windows
2023/07/28 05:27:43: Plugged in
2023/07/28 05:27:46: Unplugged
2023/07/28 05:28:06: Plugged in
2023/07/28 05:28:11: Opened serial port
2023/07/28 05:28:11: MultiWii API version received: 1.42.0
2023/07/28 05:28:11: Flight controller info, identifier: ARDU version: 3.3.1
2023/07/28 05:28:11: Running firmware released on: Mar 16 2018 08:42:29
2023/07/28 05:28:11: Board: ARCH, version: 0
2023/07/28 05:28:11: Unique device ID received - 0x01b00283056500f20363547
2023/07/28 05:28:13: MSP command failed: Offset is outside the bounds of the DataView
2023/07/28 05:28:14: MSP command failed: Offset is outside the bounds of the DataView
2023/07/28 05:28:15: MSP command failed: Offset is outside the bounds of the DataView
2023/07/28 05:28:16: MSP command failed: Offset is outside the bounds of the DataView

stylesuxx, 2023-07-28:
Yeah, that MSP is too old. You'll need to upgrade your flight-controller firmware to something more recent.

Teklock, 2023-07-28:
ArduPlane V4.5.0-dev (dd325a7a)

You can't get more current

Teklock, 2023-07-28:
Not sure why your log is showing that.

stylesuxx, 2023-07-28:
because the flight controller is returning that. The configurator is not making up such details on the fly ;-)

Teklock, 2023-07-28:
I can give you all the logs you want if you like.  I've never seen what you are showing.

stylesuxx, 2023-07-28:
You posted it yourself in the first message.

Teklock, 2023-07-28:
I just posted the only logs this interface gives me, but I'm not running anything close to that.

stylesuxx, 2023-07-28:
yes, and that info is gathered through the flight controller. First commands sent to the FC are commands to query FC firmware details and that's what's being returned.

You can also see MSP version is 1.42 and that certain MSP commands are failing. This is due to it being an old version. If you insist on running such an old version, backup your settings, flash a recent version of BF, adjust the ESCs and flash back the old version.

Teklock, 2023-07-28:
I fly quadplanes and only use Ardupilot.  Currently, I'm flying with 4.5 dev as I have told you.

7/28/2023 6:00:44 AM : PreArm: QLAND mode not armable
7/28/2023 6:00:44 AM : PreArm: Radio failsafe on
7/28/2023 6:00:30 AM : AHRS: EKF3 active
7/28/2023 6:00:30 AM : EKF3 IMU0 is using GPS
7/28/2023 6:00:13 AM : u-blox 2 HW: 00080000 SW: ROM CORE 3.01 (107888)
7/28/2023 6:00:13 AM : PreArm: QLAND mode not armable
7/28/2023 6:00:13 AM : PreArm: Radio failsafe on
7/28/2023 6:00:13 AM : PreArm: AHRS: not using configured AHRS type
7/28/2023 6:00:11 AM : GPS 2: u-blox saving config
7/28/2023 6:00:03 AM : Field Elevation Set: 187m
7/28/2023 6:00:03 AM : EKF3 IMU0 origin set
7/28/2023 5:59:57 AM : GPS 2: detected as u-blox at 230400 baud
7/28/2023 5:59:56 AM : RC Long Failsafe On: QLAND
7/28/2023 5:59:56 AM : Land complete
7/28/2023 5:59:55 AM : EKF3 IMU0 MAG0 initial yaw alignment complete
7/28/2023 5:59:54 AM : EKF3 IMU0 tilt alignment complete
7/28/2023 5:59:52 AM : EKF3 IMU0 initialised
7/28/2023 5:59:52 AM : Airspeed 1 calibrated
7/28/2023 5:59:51 AM : RC Short Failsafe: switched to QLAND
7/28/2023 5:59:51 AM : Throttle failsafe on
7/28/2023 5:59:50 AM : GPS 1: specified as DroneCAN1-125
7/28/2023 5:59:50 AM : RCOut: DS600:1-4 PWM:5-12 NeoP:13
7/28/2023 5:59:50 AM : AHRS: DCM active
7/28/2023 5:59:50 AM : ArduPilot Ready
7/28/2023 5:59:50 AM : Airspeed 1 calibration started
7/28/2023 5:59:50 AM : Barometer 2 calibration complete
7/28/2023 5:59:50 AM : Barometer 1 calibration complete
7/28/2023 5:59:49 AM : IMU0: fast sampling enabled 2.0kHz
7/28/2023 5:59:49 AM : RCOut: Initialising
7/28/2023 5:59:49 AM : MatekF765-SE 001B0028 3056500F 20363547
7/28/2023 5:59:49 AM : ChibiOS: 349ad50b
7/28/2023 5:59:49 AM : ArduPlane V4.5.0-dev (dd325a7a)
7/28/2023 5:59:49 AM : IMU0: fast sampling enabled 2.0kHz
7/28/2023 5:59:49 AM : RCOut: Initialising
7/28/2023 5:59:49 AM : MatekF765-SE 001B0028 3056500F 20363547
7/28/2023 5:59:49 AM : ChibiOS: 349ad50b
7/28/2023 5:59:49 AM : ArduPlane V4.5.0-dev (dd325a7a)
7/28/2023 5:59:49 AM : IMU0: fast sampling enabled 2.0kHz
7/28/2023 5:59:49 AM : RCOut: Initialising
7/28/2023 5:59:49 AM : MatekF765-SE 001B0028 3056500F 20363547
7/28/2023 5:59:49 AM : ChibiOS: 349ad50b
7/28/2023 5:59:49 AM : ArduPlane V4.5.0-dev (dd325a7a)
7/28/2023 5:59:48 AM : Calibrating barometer
7/28/2023 5:59:48 AM : INS: alloc 6144 bytes for ISB (free=203088)
7/28/2023 5:59:48 AM : Beginning INS calibration. Do not move plane
7/28/2023 5:59:48 AM : Ground start
7/28/2023 5:59:48 AM : QuadPlane initialised, Frame: TRI
7/28/2023 5:59:48 AM : Initialising ArduPilot
7/28/2023 5:59:48 AM : RCInput: decoding FPORT(3)

Teklock, 2023-07-28:
Like I said earlier it ended up finally working, so I'm good.

Teklock, 2023-07-28:
Is there an interest in getting https://esc-configurator.com reporting and working correctly with Ardupilot (ArduPlane, ArduCopter, etc)?

stylesuxx, 2023-07-29:
Of course. We had a couple of ardupilot users in the past check in on discord for different reasons, so I don't think it's a principle problem. Since you got it working eventually too it sounds like it's going to be a bitch to debug though.

It would be great if you had a way of reproducing the issue consistently, then I am sure we can solve it.

Can you tell me which flight controller you are using exactly? Also feel free to hop on Discord, communication might be quicker that way.

Teklock, 2023-07-30:
The FC is a Matek F765-WSE http://www.mateksys.com/?portfolio=f765-wse running https://firmware.ardupilot.org/Plane/latest/MatekF765-SE/arduplane_with_bl.hex  Sure I can I can hop on Discord.

Teklock, 2023-07-30:
When it does decide to work this is the log:

2023/07/30 01:38:21: Reader failed NetworkError: The device has been lost.
2023/07/30 01:38:21: Unplugged
2023/07/30 01:38:21: TX: cmd_InterfaceExit - 0x0 
2023/07/30 01:38:25: Plugged in
2023/07/30 01:38:33: Edge (115.0.1901.188) on Windows
2023/07/30 01:38:33: Plugged in
2023/07/30 01:38:36: Opened serial port
2023/07/30 01:38:36: MultiWii API version received: 1.42.0
2023/07/30 01:38:36: Flight controller info, identifier: ARDU version: 3.3.1
2023/07/30 01:38:36: Running firmware released on: Mar 16 2018 08:42:29
2023/07/30 01:38:36: Board: ARCH, version: 0
2023/07/30 01:38:36: Unique device ID received - 0x01b00283056500f20363547
2023/07/30 01:38:37: MSP command failed: Offset is outside the bounds of the DataView
2023/07/30 01:38:38: MSP command failed: Offset is outside the bounds of the DataView
2023/07/30 01:38:39: MSP command failed: Offset is outside the bounds of the DataView
2023/07/30 01:38:40: MSP command failed: Offset is outside the bounds of the DataView
2023/07/30 01:38:42: TX: cmd_DeviceInitFlash - 0x0 
2023/07/30 01:38:42: Trying to read 3 ESCs
2023/07/30 01:38:43: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2023/07/30 01:38:43: RX: cmd_InterfaceTestAlive - 0x0 
2023/07/30 01:38:43: ESC 1 read settings failed Unknown platform: Neither SiLabs nor Arm UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm
2023/07/30 01:38:43: Error: UnknownPlatformError: Unknown platform: Neither SiLabs nor Arm 
2023/07/30 01:38:43: TX: cmd_DeviceInitFlash - 0x1 
2023/07/30 01:38:43: Failed reading ESC 1
2023/07/30 01:38:43: RX: cmd_DeviceInitFlash - 0xB2,0xE8,0x64,0x1 
2023/07/30 01:38:43: TX: cmd_DeviceRead @ 0x1A00 - 0x70 
2023/07/30 01:38:43: RX: cmd_DeviceRead @ 0x1A00 - 0x0,0x14,0xCF,0xFF,0x33,0x1,0x1,0x19,0xFF,0x9,0x30,0x1,0xFF,0x55,0xAA,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x4,0xFF,0xFF,0xFF,0xFF,0xFF,0x28,0x50,0x4,0xFF,0x2,0xFF,0xFF,0xFF,0x0,0xFF,0xFF,0xFF,0x1,0x3F,0x2,0x0,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x23,0x51,0x5F,0x48,0x5F,0x34,0x30,0x23,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x23,0x42,0x4C,0x48,0x45,0x4C,0x49,0x24,0x45,0x46,0x4D,0x38,0x42,0x32,0x31,0x23,0x42,0x6C,0x75,0x65,0x6A,0x61,0x79,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20 
2023/07/30 01:38:43: TX: cmd_DeviceRead @ 0x1A00 - 0xFF 
2023/07/30 01:38:43: RX: cmd_DeviceRead @ 0x1A00 - 0xAB 
2023/07/30 01:38:43: TX: cmd_DeviceRead @ 0x80 - 0x80 
2023/07/30 01:38:43: RX: cmd_DeviceRead @ 0x80 - 0x0 
2023/07/30 01:38:43: ESC 2 read settings failed Cannot read properties of undefined (reading 'toString') TypeError: Cannot read properties of undefined (reading 'toString')
2023/07/30 01:38:43: Error: TypeError: Cannot read properties of undefined (reading 'toString') 
2023/07/30 01:38:43: TX: cmd_DeviceInitFlash - 0x2 
2023/07/30 01:38:43: Failed reading ESC 2
2023/07/30 01:38:43: RX: cmd_DeviceInitFlash - 0xB2,0xE8,0x64,0x1 
2023/07/30 01:38:43: TX: cmd_DeviceRead @ 0x1A00 - 0x70 
2023/07/30 01:38:43: RX: cmd_DeviceRead @ 0x1A00 - 0x0,0x14,0xCF,0xFF,0x33,0x1,0x1,0x19,0xFF,0x9,0x30,0x1,0xFF,0x55,0xAA,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x4,0xFF,0xFF,0xFF,0xFF,0xFF,0x28,0x50,0x4,0xFF,0x2,0xFF,0xFF,0xFF,0x0,0xFF,0xFF,0xFF,0x1,0x3F,0x2,0x0,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x23,0x4A,0x5F,0x48,0x5F,0x33,0x30,0x23,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x23,0x42,0x4C,0x48,0x45,0x4C,0x49,0x24,0x45,0x46,0x4D,0x38,0x42,0x32,0x31,0x23,0x42,0x6C,0x75,0x65,0x6A,0x61,0x79,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20 
2023/07/30 01:38:43: TX: cmd_DeviceRead @ 0x1A00 - 0xFF 
2023/07/30 01:38:43: RX: cmd_DeviceRead @ 0x1A00 - 0x0,0x14,0xCF,0xFF,0x33,0x1,0x1,0x19,0xFF,0x9,0x30,0x1,0xFF,0x55,0xAA,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x4,0xFF,0xFF,0xFF,0xFF,0xFF,0x28,0x50,0x4,0xFF,0x2,0xFF,0xFF,0xFF,0x0,0xFF,0xFF,0xFF,0x1,0x3F,0x2,0x0,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x23,0x4A,0x5F,0x48,0x5F,0x33,0x30,0x23,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x23,0x42,0x4C,0x48,0x45,0x4C,0x49,0x24,0x45,0x46,0x4D,0x38,0x42,0x32,0x31,0x23,0x42,0x6C,0x75,0x65,0x6A,0x61,0x79,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x2,0x3A,0x4,0x20,0x34,0x42,0xD,0x0,0x45,0x2D,0xD,0x0,0x34,0x42,0xD,0x0,0x4E,0x27,0xD3,0x0,0x45,0x2D,0xD0,0x19,0x34,0x19,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF 
2023/07/30 01:38:44: Read ESC 3: J-H-30 - Bluejay, 0.20, 48kHz
2023/07/30 01:38:44: Done reading ESCs
2023/07/30 01:39:10: TX: cmd_DeviceInitFlash - 0x0 
2023/07/30 01:39:10: Trying to read 3 ESCs
2023/07/30 01:39:10: RX: cmd_DeviceInitFlash - 0xB2,0xE8,0x64,0x1 
2023/07/30 01:39:10: TX: cmd_DeviceRead @ 0x1A00 - 0x70 
2023/07/30 01:39:10: RX: cmd_DeviceRead @ 0x1A00 - 0x0,0x14,0xCF,0xFF,0x33,0x1,0x1,0x19,0xFF,0x9,0x30,0x2,0xFF,0x55,0xAA,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x4,0xFF,0xFF,0xFF,0xFF,0xFF,0x28,0x50,0x4,0xFF,0x2,0xFF,0xFF,0xFF,0x0,0xFF,0xFF,0xFF,0x1,0x3F,0x2,0x0,0xFF,0xFF,0xFF,0xFF,0xFF, [... cut ...] x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF 
2023/07/30 01:39:11: Read ESC 3: J-H-30 - Bluejay, 0.20, 48kHz
2023/07/30 01:39:11: Done reading ESCs

stylesuxx, 2023-07-31:
Have you tried different USB cables/ports/computers - this to me seems like an issue during data transmission.

Teklock, 2023-08-01:
Yeah, I tied but it didn't work any better.  You can go ahead and close this because I'm not sure how I can give you any more information.  If I have any more issues I will reach back out again.  Thank you for the help!

stylesuxx, 2024-08-12:
Closing for now. Could not reproduce.

closed 2024-08-12 by stylesuxx