stylesuxx/esc-configurator #405: Sweet child of mine melody is off

skrobo, 2023-08-09:
I have the correct notes below... Feature Request: There should be a place where the community can upload to melody editor or at least a link in esc configurator to community posts.

SweetChi:d=4,o=5,b=125:8c#,8c#6,8g#,8f#,8f#6,8g#,8f6,8g#,8c#,8c#6,8g#,8f#,8f#6,8g#,8f6,8g#

stylesuxx, 2023-08-09:
Thank you.

Well, the place is here, you think a link from the melody to here is really necessary?

skrobo, 2023-08-09:
I am just giving some constructive criticism. It doesn’t seem right to submit an issue on github for a feature request. An upload button on melody editor along with a local save button would be ideal. Does the current save button upload or just save locally?

stylesuxx, 2023-08-09:
Yeah, I can't have people upload untested stuff. Save button saves locally.

There literally is a pinned issue to submit melodies which will be added to the configurator: https://github.com/stylesuxx/esc-configurator/issues/94

closed 2023-11-20 by stylesuxx