stylesuxx/esc-configurator #414: ECS Configurator not connecting to ECS

dareksilver, 2023-09-17:
ECS in use: HSLRC 30a 
Browser: Edge

I've managed to upload and update the firmware with blheli suite, but was looking to change the firmware to BlueJay.  Attempted to use both the browser and the standalone software, to no avail to having it connect tot he ECS (it still connects with the blheli suite with no issue.)

This is the only thing the log says (I've even gone so far as to reboot the entire PC to see if that was a issue, to receive the same errors)

-----------------------------------

2023-09-17 @ 14:55:01 -- Port already in use by another application - try re-connecting
2023-09-17 @ 14:55:01 -- Failed resetting ESC 1 - power cycle and try again
2023-09-17 @ 14:54:53 -- Opened serial port
2023-09-17 @ 14:54:50 -- Port selected
2023-09-17 @ 14:54:37 -- Plugged in
2023-09-17 @ 14:54:37 -- Edge (117.0.2045.31) on Windows

------------------------------------

I'm leaning to maybe the ECS isn't supported - but not sure.  Any help would be appreciated.

stylesuxx, 2023-09-17:
Make sure that every program that could be claiming the serial device is closed. Cura is for example known to claim serial devices. Make sure that other configurators and blheli suite is closed.

dareksilver, 2023-09-18:
> Make sure that every program that could be claiming the serial device is closed. Cura is for example known to claim serial devices. Make sure that other configurators and blheli suite is closed.

I've rebooted the PC several times, closed every program that could claim a COM port.  Even after failing to connect with the standalone bluejay configurator and the web version, I can load blheli suite and connect to the device no problem.

stylesuxx, 2023-09-18:
Which FC are you using?

dareksilver, 2023-09-18:
I'm using a Arduino Nano v3.

It's not precisely for a drone, I'm working on building a brushless foam blaster, and wanted the extra features offered by BlueJay in the ESC compared to the stock BLheli firmware.

If it helps, these are the ESC I am using in the build: https://www.amazon.com/gp/product/B07GTHDGCK/

stylesuxx, 2023-09-18:
Yeah, that's not going to work. Only Betaflight pass through mode is supported currently.

closed 2023-09-18 by stylesuxx

dareksilver, 2023-09-18:
Ah, that would explain the issues then.  Thank you for the help.

stylesuxx, 2023-09-18:
No worries. Also for foam blasters - I assume that's NERF like? Bluejay will not be the right choice. There are a couple security features in place that will only make the Motor spin up after some time. Also only DSHOT is supported.