stylesuxx/esc-configurator #416: JHEMCHU F411ELVTX 1S-2S ELRS 12A ESC 400MW VTX ESC ISSUE

jasflyfpv, 2023-09-21:
I'm having issue with my AIO flight controller. I flashed the ESC with Bluejay firmware v.16 48khz. When i arm the motors does not spin immediately and does not spin at the same time. Some does not spin at all. When i punch throtlle all motors spin but upon take off quad  flies shaking and desyncs. I copied the debug log. Hope anyone can help with this issue.
[esc-configurator-log (1).txt](https://github.com/stylesuxx/esc-configurator/files/12685900/esc-configurator-log.1.txt)
[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/12685901/esc-configurator-log.txt)

damosvil, 2023-09-21:
Please, provide photo of the build, motors info (size and kv), a video of a flight, Betaflight version, diff all and blackbox, Bluejay version and a capture with the settings.

stylesuxx, 2023-09-21:
This is not an issue with the Configurator. I'll leave this open for now. In the future if you suspect that there is an issue with Bluejay, please post in the [correct repo](https://github.com/bird-sanctuary/bluejay/issues).

stylesuxx, 2023-09-30:
@jasflyfpv - any update on this, otherwise I will close this issue.

stylesuxx, 2023-10-29:
Closed du to lack of interaction.

closed 2023-10-29 by stylesuxx