stylesuxx/esc-configurator #417: esc not detected

alexovitch16, 2023-09-30:
HI!

 after a failed flash on esc configurator esc1 doesn't appear in esc configuration website and standalone app. I' ve tried on BLHeli suite and others forks of BLheli and flash Esc 1but the flash is impossible...Have you got any solution for that?
thank you,

here is esc-configurator message:

The amount of available ESCs does not match what the flight controller is reporting. This can have multiple reasons:
The ESC was not ready yet. This can for example happen if it is playing a melody. Also a previous flash could have gone wrong and it needs longer to boot up than usual.
There are simply not more ESCs connected.
The MCU of the ESC is defective.
Try a different USB cable - sometimes they are not properly shielded and might result in such errors.
Make sure your radio is turned off while interacting with the configurator. Switched on radios have been reported to cause issues with electro-magnetic interference and thus failed read/writes.
If the result is not what you are expecting, try clicking the 'Read Settings' button again.

stylesuxx, 2023-09-30:
If the ESC does not show up, you will have to recover via C2 interface, see here: https://github.com/bird-sanctuary/bluejay/wiki/Unbrick-ESC

alexovitch16, 2023-09-30:
thanks for the reply, I was afraid about that solution because my FC is the f4 1s 5a v2.2 which no has c2 pad... After I post my question I find that post with the same FC, see the last post, it could be tricky  😅 https://github.com/jflight-public/jesc-configurator/issues/25

Thanks you !

stylesuxx, 2023-10-01:
Oh yeah - with that one it's going to get tricky I am afraid.

Good Luck!

closed 2023-10-01 by stylesuxx

Wasee045, 2024-11-13:
> HI!
>
> after a failed flash on esc configurator esc1 doesn't appear in esc configuration website and standalone app. I' ve tried on BLHeli suite and others forks of BLheli and flash Esc 1but the flash [...]

Have you got solution