stylesuxx/esc-configurator #418: SpeedyBee BLS-50A Blueay cant reverse engine

szuwar88, 2023-10-10:
As in subject. I can't reverse engine with https://esc-configurator.com/ flashing with 0.19 on SpeedyBee BLS-50A Blueay.
Thanks for help.

stylesuxx, 2023-10-10:
More info please. What are you doing exactly? You do click "write settings" after changing the direction, right?

Please post the log.

szuwar88, 2023-10-10:
I was trying flash on one engine and i was alsow trying to flash all. Also when i was trying to flash esc engines to revers on bl heli at web site its not change engine direction. 
But when i try at bl heli configurator on windows its do the job. Its only problem to change engine direction.
Also when i do it in betaflitght it also not work. 
Only one works is bl heli windows configurator. 
Thanks for help and im trying to be helpfull.

szuwar88, 2023-10-10:
bl Heli configurator change direction.
https://esc-configurator.com/ just upgrade firmware without changing direction. As in  bl heli aso in blujay.

stylesuxx, 2023-10-11:
Sorry, I can't  follow you. Please share the log from esc-configurator.com

szuwar88, 2023-10-11:
Web esc configurator dont change motor direction. After flashing IT always
change to normal. Even if u set reveresed...

stylesuxx, 2023-10-11:
Yeah, you are probably doing it wrong. You change direction and then you click "write settings". Not flash.

And one more time: Please post the log - bottom left when connected, you can download the log.

szuwar88, 2023-10-11:
I did it now. My ESC 4 is on bluejay but not reversed

2023/09/22 17:11:12: Chrome (116.0.0.0) on Windows
2023/10/04 16:42:47: Chrome (116.0.0.0) on Windows
2023/10/04 16:47:09: Port selected
2023/10/04 16:47:30: Opened serial port
2023/10/04 16:47:30: MultiWii API version received: 1.45.0
2023/10/04 16:47:30: Flight controller info, identifier: BTFL version: 4.4.2
2023/10/04 16:47:30: Running firmware released on: Jun  1 2023 02:11:42
2023/10/04 16:47:30: Board: S7X2, version: 0
2023/10/04 16:47:30: Unique device ID received - 0x03f00443232510538393431
2023/10/04 16:50:54: Trying to read 4 ESCs
2023/10/04 16:50:55: Read ESC 1: J-H-50 - BLHeli_S, 16.7
2023/10/04 16:50:56: Read ESC 2: J-H-50 - BLHeli_S, 16.7
2023/10/04 16:50:57: Read ESC 3: J-H-50 - BLHeli_S, 16.7
2023/10/04 16:50:58: Read ESC 4: J-H-50 - BLHeli_S, 16.7
2023/10/04 16:50:58: Done reading ESCs
2023/10/04 16:53:02: Chosen firmware: https://raw.githubusercontent.com/bitdump/BLHeli/master/BLHeli_S SiLabs/Hex files/J_H_50_REV16_7.HEX 
2023/10/04 16:53:04: Flashing ESC 3
2023/10/04 16:53:04: ### Step 1: Erasing EEPROM and writing safeguards 
2023/10/04 16:53:05: ### Step 2: Writing bootloader failsafe (brick protection) 
2023/10/04 16:53:05: ### Step 3: Erase, write and verify Pages 0x02-0x0D 
2023/10/04 16:53:13: ### Step 4: Erase, write and verify Pages 0x00-0x02 
2023/10/04 16:53:14: ### Step 5: Write EEPROM section 
2023/10/04 16:53:15: Flashed ESC 3 - 10.5s
2023/10/04 16:53:16: BLHELI layout found 
2023/10/04 16:53:16: Migrated setting STARTUP_POWER 
2023/10/04 16:53:16: Migrated setting MOTOR_DIRECTION 
2023/10/04 16:53:16: Migrated setting PROGRAMMING_BY_TX 
2023/10/04 16:53:16: Migrated setting COMMUTATION_TIMING 
2023/10/04 16:53:16: Migrated setting PPM_MIN_THROTTLE 
2023/10/04 16:53:16: Migrated setting PPM_MAX_THROTTLE 
2023/10/04 16:53:16: Migrated setting BEEP_STRENGTH 
2023/10/04 16:53:16: Migrated setting BEACON_STRENGTH 
2023/10/04 16:53:16: Migrated setting BEACON_DELAY 
2023/10/04 16:53:16: Migrated setting DEMAG_COMPENSATION 
2023/10/04 16:53:16: Migrated setting PPM_CENTER_THROTTLE 
2023/10/04 16:53:16: Migrated setting TEMPERATURE_PROTECTION 
2023/10/04 16:53:16: Migrated setting LOW_RPM_POWER_PROTECTION 
2023/10/04 16:53:16: Migrated setting BRAKE_ON_STOP 
2023/10/04 16:53:16: Migrated setting LED_CONTROL 
2023/10/04 16:53:16: No changes - not updating ESC 3
2023/10/04 16:53:47: Chosen firmware: https://raw.githubusercontent.com/bitdump/BLHeli/master/BLHeli_S SiLabs/Hex files/J_H_50_REV16_7.HEX 
2023/10/04 16:53:48: Flashing ESC 4
2023/10/04 16:53:49: ### Step 1: Erasing EEPROM and writing safeguards 
2023/10/04 16:53:49: ### Step 2: Writing bootloader failsafe (brick protection) 
2023/10/04 16:53:49: ### Step 3: Erase, write and verify Pages 0x02-0x0D 
2023/10/04 16:53:57: ### Step 4: Erase, write and verify Pages 0x00-0x02 
2023/10/04 16:53:58: ### Step 5: Write EEPROM section 
2023/10/04 16:53:59: Flashed ESC 4 - 10.6s
2023/10/04 16:54:00: BLHELI layout found 
2023/10/04 16:54:00: Migrated setting STARTUP_POWER 
2023/10/04 16:54:00: Migrated setting MOTOR_DIRECTION 
2023/10/04 16:54:00: Migrated setting PROGRAMMING_BY_TX 
2023/10/04 16:54:00: Migrated setting COMMUTATION_TIMING 
2023/10/04 16:54:00: Migrated setting PPM_MIN_THROTTLE 
2023/10/04 16:54:00: Migrated setting PPM_MAX_THROTTLE 
2023/10/04 16:54:00: Migrated setting BEEP_STRENGTH 
2023/10/04 16:54:00: Migrated setting BEACON_STRENGTH 
2023/10/04 16:54:00: Migrated setting BEACON_DELAY 
2023/10/04 16:54:00: Migrated setting DEMAG_COMPENSATION 
2023/10/04 16:54:00: Migrated setting PPM_CENTER_THROTTLE 
2023/10/04 16:54:00: Migrated setting TEMPERATURE_PROTECTION 
2023/10/04 16:54:00: Migrated setting LOW_RPM_POWER_PROTECTION 
2023/10/04 16:54:00: Migrated setting BRAKE_ON_STOP 
2023/10/04 16:54:00: Migrated setting LED_CONTROL 
2023/10/04 16:54:00: No changes - not updating ESC 4
2023/10/04 16:54:39: Chosen firmware: https://raw.githubusercontent.com/bitdump/BLHeli/master/BLHeli_S SiLabs/Hex files/J_H_50_REV16_7.HEX 
2023/10/04 16:54:40: Flashing ESC 1
2023/10/04 16:54:40: ### Step 1: Erasing EEPROM and writing safeguards 
[... 1041 lines cut ...]
2023/10/10 04:22:01: Migrated setting COMMUTATION_TIMING 
2023/10/10 04:22:01: Migrated setting BEEP_STRENGTH 
2023/10/10 04:22:01: Migrated setting BEACON_STRENGTH 
2023/10/10 04:22:02: Migrated setting BEACON_DELAY 
2023/10/10 04:22:02: Migrated setting DEMAG_COMPENSATION 
2023/10/10 04:22:02: Migrated setting TEMPERATURE_PROTECTION 
2023/10/10 04:22:02: Migrated setting BRAKE_ON_STOP 
2023/10/10 04:22:02: Migrated setting LED_CONTROL 
2023/10/10 04:22:02: Migrated setting POWER_RATING 
2023/10/10 04:22:02: Migrated setting STARTUP_MELODY 
2023/10/10 04:22:02: Updating ESC 4 - finished
2023/10/10 04:28:48: Reader failed NetworkError: The device has been lost.
2023/10/10 04:28:48: Unplugged
2023/10/10 04:28:59: Plugged in
2023/10/10 04:29:27: Unplugged
2023/10/10 04:29:30: Plugged in
2023/10/10 04:30:26: Unplugged
2023/10/10 04:30:28: Plugged in
2023/10/10 04:31:15: Unplugged
2023/10/10 04:35:10: Plugged in
2023/10/10 04:39:11: Opened serial port
2023/10/10 04:39:11: MultiWii API version received: 1.45.0
2023/10/10 04:39:11: Flight controller info, identifier: BTFL version: 4.4.2
2023/10/10 04:39:11: Running firmware released on: Jun  1 2023 02:11:42
2023/10/10 04:39:11: Board: S7X

szuwar88, 2023-10-11:
[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/12863236/esc-configurator-log.txt)

szuwar88, 2023-10-11:
sorry for mess

stylesuxx, 2023-10-11:
Yeah,  log clearly shows: you are FLASHING. You are not writing the settings. You don't need to press "Flash" or "Flash all". Just change the direction in the dropdown and click "write settings" on the bottom right.

szuwar88, 2023-10-11:
amazing

closed 2023-10-11 by stylesuxx

szuwar88, 2023-10-11:
can we talk on priv?

stylesuxx, 2023-10-11:
You  can join our Discord: https://discord.gg/QvSS5dk23C