stylesuxx/esc-configurator #421: Ukrainian translation

olegshevtsov, 2023-11-11:
**Is your feature request related to a problem? Please describe.**
There is a big community of Ukrainian users. We would like to make a Ukrainian version of the interface for them.

**Describe the solution you'd like**
We have translated Betaflight Configurator, and ELRS Configurator. Time to translate ESC Configurator.

**Describe alternatives you've considered**
ESC Configurator is quite simple and easy to understand even in English. So not translating is also an option. People could use Google Tranlate.

**Additional context**
Thanks for your work, quite many people use your software on a daily basis.

stylesuxx, 2023-11-12:
Hey, Ukrainian would be very much appreciated - please [ping me on crowdin](https://crowdin.com/project/esc-configuratorcom) and I will give you permissions for Ukrainian. I already added the language, I would just give you proof reader permissions.

stylesuxx, 2023-11-20:
@olegshevtsov Please also let me know which string I should use for "Ukrainian" in the drop down menu, that is not part of the translations and I need to add it manually.

olegshevtsov, 2023-11-20:
It is 
 Українська - if you need it capitalized
 українська - without capitalization

stylesuxx, 2023-11-28:
Translation merged, added language to dropdown.

closed 2023-11-28 by stylesuxx